node-usb/node-usb #66: Compile problem

kylepaulsen, 2015-02-24:
Windows 7 64 bit with 64 bit version of node 0.12.0
I am getting compile errors like this:

c:\users\kyle\desktop\step_tracker\node_modules\usb\src\protobuilder.h(38): error C2061: syntax error : identifier 'InvocationCallback' (..\src\node_usb.cc) [c
:\Users\Kyle\Desktop\step_tracker\node_modules\usb\build\usb_bindings.vcxproj]
c:\users\kyle\desktop\step_tracker\node_modules\usb\src\protobuilder.h(55): error C2039: 'InvocationCallback' : is not a member of 'v8' (..\src\node_usb.cc) [c
:\Users\Kyle\Desktop\step_tracker\node_modules\usb\build\usb_bindings.vcxproj]
c:\users\kyle\desktop\step_tracker\node_modules\usb\src\protobuilder.h(55): error C2061: syntax error : identifier 'InvocationCallback' (..\src\node_usb.cc) [c
:\Users\Kyle\Desktop\step_tracker\node_modules\usb\build\usb_bindings.vcxproj]
c:\users\kyle\desktop\step_tracker\node_modules\usb\src\protobuilder.h(59): error C2039: 'InvocationCallback' : is not a member of 'v8' (..\src\node_usb.cc) [c
:\Users\Kyle\Desktop\step_tracker\node_modules\usb\build\usb_bindings.vcxproj]
c:\users\kyle\desktop\step_tracker\node_modules\usb\src\protobuilder.h(59): error C2061: syntax error : identifier 'InvocationCallback' (..\src\node_usb.cc) [c
:\Users\Kyle\Desktop\step_tracker\node_modules\usb\build\usb_bindings.vcxproj]
c:\users\kyle\desktop\step_tracker\node_modules\usb\src\protobuilder.h(63): error C2061: syntax error : identifier 'AccessorGetter' (..\src\node_usb.cc) [c:\Us
ers\Kyle\Desktop\step_tracker\node_modules\usb\build\usb_bindings.vcxproj]
c:\users\kyle\desktop\step_tracker\node_modules\usb\src\protobuilder.h(34): error C2660: 'v8::FunctionTemplate::New' : function does not take 0 arguments (..\s
rc\node_usb.cc) [c:\Users\Kyle\Desktop\step_tracker\node_modules\usb\build\usb_bindings.vcxproj]
c:\users\kyle\desktop\step_tracker\node_modules\usb\src\protobuilder.h(34): error C2660: 'v8::PersistentBase<T>::New' : function does not take 1 arguments [c:\
Users\Kyle\Desktop\step_tracker\node_modules\usb\build\usb_bindings.vcxproj]

I have followed the instructions on the readme. I have libusbx here: C:\Program Files\libusb
I also used zadig for good measure, even though I think that's only needed after the module installs.

Please Help?

kevinmehall, 2015-02-24:
What version of node-usb are you running? Only versions >1.0 are compatible with node 0.12, and should download binaries via node-shyp instead of building.

kylepaulsen, 2015-02-26:
Sorry, I copied a project with an old package.json (with an old version of node-usb). Thanks for your help.

closed 2015-02-26 by kylepaulsen