node-usb/node-usb #68: Atom-Shell

Trixt0r, 2015-02-26:
Hi,

when I try to build the module for atom-shell 0.21.0 I get the following errors:

```
gyp info it worked if it ends with ok
gyp info using node-gyp@1.0.2
gyp info using node@0.12.0 | darwin | x64
child_process: customFds option is deprecated, use stdio instead.
gyp info spawn python
gyp info spawn args [ '/usr/local/lib/node_modules/node-gyp/gyp/gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'make',
gyp info spawn args   '-I',
gyp info spawn args   '/Users/reich/Desktop/dropclient/app/node_modules/dropclient-zpl/node_modules/usb/build/config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/usr/local/lib/node_modules/node-gyp/addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/Users/reich/.atom-shell-gyp/.node-gyp/0.21.0/common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=/Users/reich/.atom-shell-gyp/.node-gyp/0.21.0',
gyp info spawn args   '-Dmodule_root_dir=/Users/reich/Desktop/dropclient/app/node_modules/dropclient-zpl/node_modules/usb',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'build',
gyp info spawn args   '-Goutput_dir=.' ]
gyp info spawn make
gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
  CC(target) Release/obj.target/libusb/libusb/libusb/core.o
../libusb/libusb/core.c:2241:35: warning: comparison of integers of different signs: 'int' and 'unsigned long' [-Wsign-compare]
        if (header_len < 0 || header_len >= sizeof(buf)) {
                              ~~~~~~~~~~ ^  ~~~~~~~~~~~
../libusb/libusb/core.c:2250:44: warning: comparison of integers of different signs: 'int' and 'unsigned long' [-Wsign-compare]
        if (text_len < 0 || text_len + header_len >= sizeof(buf)) {
                            ~~~~~~~~~~~~~~~~~~~~~ ^  ~~~~~~~~~~~
2 warnings generated.
  CC(target) Release/obj.target/libusb/libusb/libusb/descriptor.o
  CC(target) Release/obj.target/libusb/libusb/libusb/hotplug.o
  CC(target) Release/obj.target/libusb/libusb/libusb/io.o
  CC(target) Release/obj.target/libusb/libusb/libusb/strerror.o
  CC(target) Release/obj.target/libusb/libusb/libusb/sync.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/poll_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/threads_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/darwin_usb.o
../libusb/libusb/os/darwin_usb.c:1281:27: warning: expression which evaluates to zero treated as a null pointer constant of type 'IOUSBInterfaceInterface550 **'
      (aka 'struct IOUSBInterfaceStruct550 **') [-Wnon-literal-null-conversion]
  cInterface->interface = IO_OBJECT_NULL;
                          ^~~~~~~~~~~~~~
/System/Library/Frameworks/IOKit.framework/Headers/IOTypes.h:164:24: note: expanded from macro 'IO_OBJECT_NULL'
#define IO_OBJECT_NULL  ((io_object_t) 0)
                        ^~~~~~~~~~~~~~~~~
1 warning generated.
  LIBTOOL-STATIC Release/usb.a
  CXX(target) Release/obj.target/usb_bindings/src/node_usb.o
In file included from ../src/node_usb.cc:1:
In file included from ../src/node_usb.h:21:
In file included from ../src/helpers.h:3:
../node_modules/nan/nan.h:67:21: error: no type named 'ExternalAsciiStringResource' in 'v8::String'; did you mean 'ExternalStringResource'?
typedef v8::String::ExternalAsciiStringResource
        ~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
                    ExternalStringResource
/Users/reich/.atom-shell-gyp/.node-gyp/0.21.0/deps/v8/include/v8.h:1936:19: note: 'ExternalStringResource' declared here
  class V8_EXPORT ExternalStringResource
                  ^
In file included from ../src/node_usb.cc:1:
In file included from ../src/node_usb.h:21:
In file included from ../src/helpers.h:3:
[... 74 lines cut ...]
/Users/reich/.atom-shell-gyp/.node-gyp/0.21.0/deps/v8/include/v8config.h:329:45: note: expanded from macro 'V8_DEPRECATED'
# define V8_DEPRECATED(message, declarator) declarator
                                            ^
9 errors generated.
make: *** [Release/obj.target/usb_bindings/src/node_usb.o] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1067:12)
gyp ERR! System Darwin 13.4.0
gyp ERR! command "node" "/usr/local/bin/node-gyp" "rebuild" "--target=0.21.0" "--dist-url=https://atom.io/download/atom-shell"
gyp ERR! cwd /Users/reich/Desktop/dropclient/app/node_modules/dropclient-zpl/node_modules/usb
gyp ERR! node -v v0.12.0
gyp ERR! node-gyp -v v1.0.2
gyp ERR! not ok 
```

I am building it with atom-shell headers:

`HOME=~/.atom-shell-gyp node-gyp rebuild --target=0.21.0 --dist-url=https://atom.io/download/atom-shell`

Can someone tell me how to solve this problem?

-Trixt0r

kevinmehall, 2015-02-26:
https://github.com/rvagg/nan/commit/0b7f1bb505616cae9a1a89f93e6485b7a4fbced0 is not yet in a `nan` release. You may be able to replace `node_modules/usb/node_modules/nan` with a git checkout or `npm link` of `nan` from git.

kevinmehall, 2015-02-28:
`nan` 1.7.0 is out and claims support for atom-shell 0.21, so you should be good to go. Feel free to reopen if you encounter a problem specific to node-usb.

closed 2015-02-28 by kevinmehall