node-usb/node-usb #69: switch to async functions

jacobrosenthal, 2015-02-28:
Greetings, Im a maintainer of browser-serialport, a shim for chrome's serial to use the node-serialport api so we can just webpack back and forth between apis.

I was hoping to do similar for node-usb but was stopped pretty early due to the synchronous nature of open and find.

Was there a reason for this decision? How would you feel about moving towards async for a) compatibility and b) to not block so long :)

Is there a reason you

jacobrosenthal, 2015-03-13:
Well for those interested, I have an [arduino programmer board](https://www.sparkfun.com/products/11801) and [associated node library](https://github.com/jacobrosenthal/usbtinyisp) that only needs controltransfer.  Ive [forked node-usb](https://github.com/jacobrosenthal/arduino-usb/tree/async) and so far just changed open and findbyids slightly and now I have a [minimally implemented chrome shim working in browser](https://github.com/jacobrosenthal/browser-usb)

kevinmehall, 2015-10-27:
Opening and finding devices does not perform any IO. It only fetches data already cached in RAM in the kernel. So there's no technical reason to make them async, besides compatibility with browser APIs.

I'm not about to make a breaking change, but I'll leave this open to keep in mind for 2.0.