node-usb/node-usb #72: Fail to install on Linux

fulup-bzh, 2015-03-16:
When trying to install on linux following command fails
```
 node shyp-blacklist.js win32-x64 win32-ia32 darwin-x64 || node-gyp rebuild
```
win32-x64 win32-ia32 darwin-x64 are in optionnal depencies and obviously does not exist on linux.
I'm on OpenSuse 13.4 with everything needed to compile C/C++.

```
fulup@fulup-tizen:~/Documents/WorkSpace/ScarlettFocusrit/node_modules> npm install usb
npm WARN optional dep failed, continuing usb-shyp-darwin-x64@1.0.6
npm WARN optional dep failed, continuing usb-shyp-win32-x64@1.0.5
npm WARN optional dep failed, continuing usb-shyp-win32-ia32@1.0.5

> usb@1.0.5 install /home/fulup/Documents/WorkSpace/ScarlettFocusrit/node_modules/usb
> node shyp-blacklist.js win32-x64 win32-ia32 darwin-x64 || node-gyp rebuild

gyp: /usr/share/node/common.gypi not found (cwd: /home/fulup/Documents/WorkSpace/ScarlettFocusrit/node_modules/usb) while reading includes of binding.gyp while trying to load binding.gyp
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/lib64/node_modules/npm/node_modules/node-gyp/lib/configure.js:338:16)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:810:12)
gyp ERR! System Linux 3.16.7-7-desktop
gyp ERR! command "node" "/usr/lib64/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/fulup/Documents/WorkSpace/ScarlettFocusrit/node_modules/usb
gyp ERR! node -v v0.10.31
gyp ERR! node-gyp -v v1.0.1
gyp ERR! not ok 
npm ERR! usb@1.0.5 install: `node shyp-blacklist.js win32-x64 win32-ia32 darwin-x64 || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the usb@1.0.5 install script.

```

kevinmehall, 2015-03-16:
The shyp-blacklist command is meant to fail on Linux, and the `node-gyp rebuild` part after the `||` executes to build it from source.

The actual error is `gyp: /usr/share/node/common.gypi not found`, which sounds like something wrong with your node or node-gyp installation.

fulup-bzh, 2015-03-16:
Exact 'sudo npm install -g node-gyp' fails to install /usr/share/node/common.gypi

Look like a common problem. https://github.com/TooTallNate/node-gyp/issues/313 I did hard copy file from an old installation and it worked, the trick about deleting old gyp package did not work.

Your dependencies list also lack nodejs-dev package. Today most people do not recompile binary packages and if "nodejs-dev" is not install it fails with V8.h missing, which is not an obvious error for everyone.

Thank you

closed 2015-03-16 by fulup-bzh