node-usb/node-usb #74: Error install usb

FlorianCcj, 2015-04-17:
I tried to install node usb it seems to succed but nothing works 

      C:\Users\Florian>npm install usb
      npm WARN optional dep failed, continuing usb-shyp-darwin-x64@1.0.6

      > usb@1.0.5 install C:\Users\Florian\node_modules\usb
      > node shyp-blacklist.js win32-x64 win32-ia32 darwin-x64 || node-gyp rebuild

      usb@1.0.5 node_modules\usb
      ├── bindings-shyp@0.2.3
      ├── nan@1.7.0
      ├── usb-shyp-win32-ia32@1.0.5
      └── usb-shyp-win32-x64@1.0.5

      C:\Users\Florian>npm ls installed
      C:\Users\Florian
      └── (empty)

      npm ERR! code 1

      C:\Users\Florian>node -v
      v0.12.2

      C:\Users\Florian>npm -v
      2.7.4

      C:\Users\Florian>

kevinmehall, 2015-04-19:
`npm ls installed` lists packages named "installed". Try `npm ls`.

closed 2015-04-19 by kevinmehall

FlorianCcj, 2015-04-20:
here is the result :
     C:\Users\Florian>npm ls
     C:\Users\Florian
     └─┬ usb@1.0.5
       ├── bindings-shyp@0.2.3
       ├── nan@1.7.0
       ├── usb-shyp-win32-ia32@1.0.5
       └── usb-shyp-win32-x64@1.0.5

     C:\Users\Florian>

And if I try just this line with nothing else
     var usb = require("usb");
Here is the result 
    "C:\Program Files (x86)\JetBrains\WebStorm 9.0.1\bin\runnerw.exe" "C:\Program                         Files\nodejs\node.exe" test.js
    module.js:338
        throw err;
              ^
    Error: Cannot find module 'usb'
        at Function.Module._resolveFilename (module.js:336:15)
        at Function.Module._load (module.js:278:25)
        at Module.require (module.js:365:17)
        at require (module.js:384:17)
        at Object.<anonymous> (c:\Document\Logiciel\WebstormProjects\M1_305\test.js:8:11)
        at Module._compile (module.js:460:26)
        at Object.Module._extensions..js (module.js:478:10)
        at Module.load (module.js:355:32)
        at Function.Module._load (module.js:310:12)
        at Function.Module.runMain (module.js:501:10)

    Process finished with exit code 1