node-usb/node-usb #80: Events on new device or removal of device

cstyan, 2015-05-22:
I'm hoping someone can help me either get this working if the functionality already exists, or point me in the right direction so I can implement the functionality myself and submit a PR.

Basically what I want to do is have some async function running in the background that emits events when a device is plugged in or when a device is unplugged.  The main portion of my program will be communicating with these devices and will need handle being notified of the events.  Currently I'm doing this via monitoring the size of the array returned by `getDeviceList` but there must be a better way.

kevinmehall, 2015-05-22:
Hotplug support exists, but needs to be added to the documentation:

```
var usb = require('usb');
usb.on('attach', function(device) { ... });
usb.on('detach', function(device) { ... });
```

cstyan, 2015-05-22:
Ah perfect, that's exactly what I needed.  Thanks.

kevinmehall, 2015-10-27:
Fixed in #84

closed 2015-10-27 by kevinmehall