node-usb/node-usb #81: How do I get mount point path ?

richardliu-ami, 2015-06-09:
Hi,

I want to use 

```javascript
usb.findByIds(vid, pid)
```
to find out the usb storage path (ex. /mnt/xxx-xxx) but I seem don't see way to do that. Can anyone give me some hint. Thanks.

angelmic, 2015-06-10:
I modified the node-usb-detection for mac, please check [this](https://github.com/angelmic/node-usb-detection/tree/dev). Only for mac now. Linux later. Windows .. unknow.

kevinmehall, 2015-06-10:
This library is for communicating with device endpoints directly. It won't help you if you want to use the OS drivers for mass storage and filesystems.

closed 2015-06-10 by kevinmehall

richardliu-ami, 2015-06-11:
I got it! Thanks!

sudhakar3697, 2020-04-03:
check [detect-usb](https://www.npmjs.com/package/detect-usb) out. Gives mount path