node-usb/node-usb #86: Wish: Add deviceName and manufacturerName to .deviceDescriptor

armaseg, 2015-08-28:
I realize that you are already implementing hotplug events ( #80 ), nevertheless I use usb-detection package [ https://github.com/MadLittleMods/node-usb-detection ] because it returns the deviceName and manufacturerName.

I think it would be great if you add these info to your deviceDescriptor.

<code>
deviceName: 'Teensy USB Serial (COM3)',
manufacturer: 'PJRC.COM, LLC.',
</code>

Thank you!

kevinmehall, 2015-10-27:
You can do this easily with the library, but I'd accept a PR that added convenience functions. Fetching the string involves IO, so it'd have to be an async function with a callback.

```
dev.getStringDescriptor(dev.deviceDescriptor.iManufacturer, cb)
dev.getStringDescriptor(dev.deviceDescriptor.iProduct, cb)
dev.getStringDescriptor(dev.deviceDescriptor.iSerialNumber, cb)
```

You should check that the descriptor index is nonzero, because zero indicates that the descriptor is not present.

Nic2012, 2016-04-09:
Hi,
is there any progress for this enhancements? I can need this too. An extra getStringDescriptor call is working fine in case of an attach event but when y detect an detach event there is no way to get deviceName any more, otherwise I have to cache names of each attached devices done before.

Best regards,
Nic

kevinmehall, 2016-04-09:
You'll have to cache them before the device is disconnected. There's no way for the library to communicate with a disconnected device either.

Nic2012, 2016-04-10:
Yes I know, but I do not want to implement extra infrastructure thats why I am using sometimes libs from npm ;)

ivankennethwang, 2018-09-18:
Hello, were you guys able to get the deviceName? I only see manufacturer, product and serial number.

ivankennethwang, 2018-09-18:
```
device.open();
  device.getStringDescriptor(deviceDescriptor.iManufacturer, function (err, manufacturer) {
    device.getStringDescriptor(deviceDescriptor.iProduct, function (err, product) {
      device.getStringDescriptor(deviceDescriptor.iSerialNumber, function (err, serialNumber) {
        console.log(manufacturer, product, serialNumber);
        device.close();
      }));
    });
  });
});

only shows the first one

ivankennethwang, 2018-09-18:
Sorry ignore the deviceName, I narrowed down the problem into only getting a response from the first one. I tried switching iManufacturer with iProduct and I am getting product string descriptor only. Not sure what I'm doing wrong here.

vervaekejonathan, 2019-01-14:
> ```
> device.open();
> device.getStringDescriptor(deviceDescriptor.iManufacturer, function (err, manufacturer) {
>   device.getStringDescriptor(deviceDescriptor.iProduct, function (err, product) { [...]

@ivankennethwang  this is because your object (device) is gone after calling the first function. When inside the first handler, device is not open anymore / does not exist