node-usb/node-usb #88: Node webkit app error not valid win32 application

shraddha1112, 2015-09-04:
Hello,
I have created a sample application using this module. When I run it as node app it lists all usb. Now when I wrap it with Node-webkit same application is crashing, As per the issue comments I have complied and build the usb module using following command:

node-pre-gyp configure --runtime=node-webkit --target=0.12.3
node-pre-gyp build --runtime=node-webkit --target=0.12.3

But I am getting following error:

"%1 is not a valid Win32 application.
↵D:\Projects\…er\node_modules\usb\src\binding\usb_bindings.node"}

changuish, 2015-09-14:
maybe you did not rebuild the source.the files you down recently are soure code,only can bu used after build

shraddha1112, 2015-09-22:
changuish: Thanks that worked!!!

closed 2015-09-22 by shraddha1112

jeniaSakirko, 2016-04-06:
How do you rebuild the source?

shraddha1112, 2018-08-06:
Hi, Actually it was long back. Probably its nw-gyp rebuild --target=0.5.0 command.