node-usb/node-usb #89: Update NAN to 2.0.x

abritinthebay, 2015-09-11:
This will add io.js and NodeJS v4.0 compatibility.

joshperry, 2015-10-15:
Hey guys, I was looking at PR #90 and #91to see if I could lend a hand landing this on node v4.

As I was going through the code I noticed some pretty serious architectural issues, the biggest of which is the misuse of libusb.

The [GetDeviceList](https://github.com/nonolith/node-usb/blob/master/src/node_usb.cc#L106) function calls [libusb_get_device_list](http://libusb.sourceforge.net/api-1.0/group__dev.html#gac0fe4b65914c5ed036e6cbec61cb0b97), builds an array of javascript objects, and then before returning them, calls [libusb_free_device_list](http://libusb.sourceforge.net/api-1.0/group__dev.html#gad3b8561d064bb3e1b8851ddeed3cd7d6) with the `unref_devices` parameter set to `true`.

In the `libusb_get_device_list` documentation it warns:

> Note that libusb_free_device_list() can unref all the devices for you. Be careful not to unreference a device you are about to open until after you have opened it.

I think the best fix for this (though it is API breaking) is to have `Device::GetDeviceList` accept a callback, call that callback with the list of devices to give the caller an opportunity to open an interesting device, and then call `libusb_free_device_list` once the user callback returns.

Does pushing an API change like this in conjunction with node v4 support sound like something users would be open to?

kevinmehall, 2015-10-15:
It creates a `Device` object for each device in the list, which increments the reference count:
https://github.com/nonolith/node-usb/blob/2f5af73629aa70c169769bdb9d9595b9d869e8ed/src/device.cc#L19

Thus when libusb_free_device_list unrefs them, the reference count drops to 1, not 0, and they are not freed.

joshperry, 2015-10-15:
@kevinmehall Ah, I see! That makes sense.

However, correct me if I'm wrong, I'm new to the codebase, but isn't that even worse, considering: https://code.google.com/p/v8-juice/wiki/CommentaryOnV8#No_GC_guaranty

What are the potential issues with potentially keeping libusb devices refed?

kevinmehall, 2015-10-15:
It's no different than JS objects -- they simply won't be freed unless GC runs. `libusb_device` is just a C data structure with info about a device. It doesn't involve any IO. If you `.open()` a device and fail to `.close()` it, yes, it will stay open until GC runs, or the process exits and the kernel cleans up after it.

joshperry, 2015-10-15:
So the ref counting of `libusb_device` instances are just to prevent leaking the structure? If that's the case, then yeah, the remote possibility of leaking a handful of those instances for the duration of a process doesn't seem like it would cause any problems.

kevinmehall, 2015-10-15:
Exactly. A `libusb_device` is likely lighter than the JS `Device` object that wraps it.

joshperry, 2015-10-15:
@kevinmehall Thanks for clarifying, kind of new here and just trying to find somewhere I can help out.

kevinmehall, 2015-10-27:
Closed in #91 / 1.1.0

closed 2015-10-27 by kevinmehall