node-usb/node-usb #98: Cannot install as root on archlinux.

RX14, 2015-10-27:
I cannot install the udb module as root:

```
$ s npm install usb -g

> usb@1.1.0 install /usr/lib/node_modules/usb
> node-pre-gyp install --fallback-to-build

gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/4.2.1"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/usr/lib/node_modules/usb/.node-gyp"
make: Entering directory '/usr/lib/node_modules/usb/build'
make: *** No rule to make target '../.node-gyp/4.2.1/include/node/common.gypi', needed by 'Makefile'.  Stop.
make: Leaving directory '/usr/lib/node_modules/usb/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:270:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Linux 4.2.3-1-ARCH
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/usr/lib/node_modules/usb/src/binding/usb_b
indings.node" "--module_name=usb_bindings" "--module_path=/usr/lib/node_modules/usb/src/binding"
gyp ERR! cwd /usr/lib/node_modules/usb
gyp ERR! node -v v4.2.1
gyp ERR! node-gyp -v v3.0.3
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute '/usr/bin/node /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/usr/lib/node_modul
es/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/usr/lib/node_modules/usb/src/binding' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/usb/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:87:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:818:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:211:5)
node-pre-gyp ERR! System Linux 4.2.3-1-ARCH
node-pre-gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/usb/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /usr/lib/node_modules/usb
node-pre-gyp ERR! node -v v4.2.1
node-pre-gyp ERR! node-pre-gyp -v v0.6.4
node-pre-gyp ERR! not ok
Failed to execute '/usr/bin/node /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/usr/lib/node_modules/usb/src/binding/usb_bindings
.node --module_name=usb_bindings --module_path=/usr/lib/node_modules/usb/src/binding' (1)
npm ERR! Linux 4.2.3-1-ARCH
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "install" "usb" "-g"
npm ERR! node v4.2.1
npm ERR! npm  v3.3.9
npm ERR! code ELIFECYCLE

npm ERR! usb@1.1.0 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the usb@1.1.0 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! This is most likely a problem with the usb package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls usb
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /data/programming/tessel-test/npm-debug.log
```

Installing works when not root.

stevelacy, 2015-10-27:
Received the same error on:
OS: Debian 8
Kernel: 4.2.0 AMD64

RX14, 2015-10-27:
Found an issue with node-gyp: nodejs/node-gyp#454

Fix: `s npm install --unsafe-perm t2-cli -g`

andreaskern, 2015-10-29:
I had the problem too
arch linux on rasperry pi as root
`npm install --unsafe-perm --global node-red` worked

kevinmehall, 2016-04-02:
This is https://github.com/nodejs/node-gyp/issues/454

My recommendation is to never run npm as root/sudo, which npm's defaults on Linux make unnecessarily difficult. Follow [these instructions](https://github.com/sindresorhus/guides/blob/master/npm-global-without-sudo.md) fix `npm -g` to install under your home directory.

closed 2016-04-02 by kevinmehall