node-usb/node-usb #103: Failing to build on Raspberry Pi

scottsweb, 2015-11-08:
I am having some issues with this package on my Pi. The output is as follows:

```
npm install usb
\
> usb@1.1.0 install /home/scott/.node-red/node_modules/usb
> node-pre-gyp install --fallback-to-build

make: Entering directory '/home/scott/.node-red/node_modules/usb/build'
  CC(target) Release/obj.target/libusb/libusb/libusb/core.o
  CC(target) Release/obj.target/libusb/libusb/libusb/descriptor.o
  CC(target) Release/obj.target/libusb/libusb/libusb/hotplug.o
  CC(target) Release/obj.target/libusb/libusb/libusb/io.o
  CC(target) Release/obj.target/libusb/libusb/libusb/strerror.o
  CC(target) Release/obj.target/libusb/libusb/libusb/sync.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/poll_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/threads_posix.o
../libusb/libusb/os/threads_posix.c:24:0: warning: "_GNU_SOURCE" redefined [enabled by default]
<command-line>:0:0: note: this is the location of the previous definition
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_usbfs.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_udev.o
  AR(target) Release/obj.target/usb.a
  COPY Release/usb.a
  CXX(target) Release/obj.target/usb_bindings/src/node_usb.o
  CXX(target) Release/obj.target/usb_bindings/src/device.o
../src/device.cc: In static member function ‘static v8::Local<v8::Object> Device::get(libusb_device*)’:
../src/device.cc:47:19: error: ‘class std::map<libusb_device*, Nan::Persistent<v8::Object> >’ has no member named ‘emplace’
../src/device.cc:47:37: error: unable to deduce ‘auto’ from ‘<expression error>’
usb_bindings.target.mk:96: recipe for target 'Release/obj.target/usb_bindings/src/device.o' failed
make: *** [Release/obj.target/usb_bindings/src/device.o] Error 1
make: Leaving directory '/home/scott/.node-red/node_modules/usb/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:269:23)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1074:12)
gyp ERR! System Linux 4.1.7-v7+
gyp ERR! command "/usr/bin/nodejs" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/home/scott/.node-red/node_modules/usb/src/binding/usb_bindings.node" "--module_name=usb_bindings" "--module_path=/home/scott/.node-red/node_modules/usb/src/binding"
gyp ERR! cwd /home/scott/.node-red/node_modules/usb
gyp ERR! node -v v0.12.7
gyp ERR! node-gyp -v v2.0.1
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/usr/bin/nodejs /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/scott/.node-red/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/home/scott/.node-red/node_modules/usb/src/binding' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/scott/.node-red/node_modules/usb/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
node-pre-gyp ERR! stack     at maybeClose (child_process.js:1015:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1087:5)
node-pre-gyp ERR! System Linux 4.1.7-v7+
node-pre-gyp ERR! command "node" "/home/scott/.node-red/node_modules/usb/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/scott/.node-red/node_modules/usb
node-pre-gyp ERR! node -v v0.12.7
node-pre-gyp ERR! node-pre-gyp -v v0.6.4
node-pre-gyp ERR! not ok 
Failed to execute '/usr/bin/nodejs /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/scott/.node-red/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/home/scott/.node-red/node_modules/usb/src/binding' (1)
npm ERR! Linux 4.1.7-v7+
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "install" "usb"
npm ERR! node v0.12.7
npm ERR! npm  v2.11.3
npm ERR! code ELIFECYCLE

npm ERR! usb@1.1.0 install: `node-pre-gyp install --fallback-to-build`

npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the usb@1.1.0 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! This is most likely a problem with the usb package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls usb
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/scott/.node-red/npm-debug.log
```

Any ideas on how to debug and fix it?

kevinmehall, 2015-11-14:
You need a version of GCC / libstdc++ that supports C++11, specifically std::map::emplace.

kevinmehall, 2015-11-14:
Try https://github.com/nonolith/node-usb/commit/dfc63ff354fa38dcfd1697bd1b65c917fc81783f

scottsweb, 2015-11-14:
Thanks for the suggestions @kevinmehall 

GCC was at version `4.6`. I just updated to `4.7.2` as [this thread](https://www.raspberrypi.org/forums/viewtopic.php?f=33&t=22938) seemed to suggest that would have C++11. I then ran: `npm install git://github.com/nonolith/node-usb.git#dfc63ff`but unfortunately it still fails. Full output below:

```
npm WARN installMany node-pre-gyp was bundled with usb@1.1.0, but bundled package wasn't found in unpacked tree
npm WARN installMany nopt was bundled with node-pre-gyp@0.6.4, but bundled package wasn't found in unpacked tree
npm WARN installMany npmlog was bundled with node-pre-gyp@0.6.4, but bundled package wasn't found in unpacked tree
npm WARN installMany request was bundled with node-pre-gyp@0.6.4, but bundled package wasn't found in unpacked tree
npm WARN installMany semver was bundled with node-pre-gyp@0.6.4, but bundled package wasn't found in unpacked tree
npm WARN installMany tar was bundled with node-pre-gyp@0.6.4, but bundled package wasn't found in unpacked tree
npm WARN installMany tar-pack was bundled with node-pre-gyp@0.6.4, but bundled package wasn't found in unpacked tree
npm WARN installMany mkdirp was bundled with node-pre-gyp@0.6.4, but bundled package wasn't found in unpacked tree
npm WARN installMany rc was bundled with node-pre-gyp@0.6.4, but bundled package wasn't found in unpacked tree
npm WARN installMany rimraf was bundled with node-pre-gyp@0.6.4, but bundled package wasn't found in unpacked tree
-
> usb@1.1.0 install /home/scott/.node-red/node_modules/usb
> node-pre-gyp install --fallback-to-build

make: Entering directory '/home/scott/.node-red/node_modules/usb/build'
make: *** No rule to make target 'Release/obj.target/libusb/libusb/libusb/core.o', needed by 'Release/obj.target/usb.a'.  Stop.
make: Leaving directory '/home/scott/.node-red/node_modules/usb/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:269:23)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1074:12)
gyp ERR! System Linux 4.1.7-v7+
gyp ERR! command "/usr/bin/nodejs" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/home/scott/.node-red/node_modules/usb/src/binding/usb_bindings.node" "--module_name=usb_bindings" "--module_path=/home/scott/.node-red/node_modules/usb/src/binding"
gyp ERR! cwd /home/scott/.node-red/node_modules/usb
gyp ERR! node -v v0.12.7
gyp ERR! node-gyp -v v2.0.1
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/usr/bin/nodejs /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/scott/.node-red/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/home/scott/.node-red/node_modules/usb/src/binding' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/scott/.node-red/node_modules/usb/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
node-pre-gyp ERR! stack     at maybeClose (child_process.js:1015:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1087:5)
node-pre-gyp ERR! System Linux 4.1.7-v7+
node-pre-gyp ERR! command "node" "/home/scott/.node-red/node_modules/usb/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/scott/.node-red/node_modules/usb
node-pre-gyp ERR! node -v v0.12.7
node-pre-gyp ERR! node-pre-gyp -v v0.6.4
node-pre-gyp ERR! not ok 
Failed to execute '/usr/bin/nodejs /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/scott/.node-red/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/home/scott/.node-red/node_modules/usb/src/binding' (1)
npm ERR! Linux 4.1.7-v7+
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "install" "git://github.com/nonolith/node-usb.git#dfc63ff"
npm ERR! node v0.12.7
npm ERR! npm  v2.11.3
npm ERR! code ELIFECYCLE

npm ERR! usb@1.1.0 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the usb@1.1.0 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! This is most likely a problem with the usb package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls usb
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/scott/.node-red/npm-debug.log
```

kevinmehall, 2015-11-14:
If you install from a `git://` url, `npm` doesn't clone the `libusb` submodule, so it fails to build. Fetch the source with `git clone --recursive https://github.com/nonolith/node-usb`, `cd` in and `npm install`.

scottsweb, 2015-11-14:
Thanks again. That seems to have built which is great news.

kevinmehall, 2015-11-14:
Merged 4a8eb67ff

closed 2015-11-14 by kevinmehall