node-usb/node-usb #104: Crashes opening device on Mac OS X Node v5.0.0

brandonros, 2015-11-12:
```
var usb = require('usb');

usb.setDebugLevel(4);

var device = usb.findByIds(0x0eb8, 0xf000);

device.open();
```

```
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 3.034133] [0000050f] libusb: debug [libusb_get_device_list]
[ 3.034177] [0000050f] libusb: debug [discovered_devs_append] need to increase capacity
[ 3.034216] [0000050f] libusb: debug [libusb_get_device_descriptor]
[ 3.034259] [0000050f] libusb: debug [libusb_get_device_descriptor]
[ 3.034278] [0000050f] libusb: debug [libusb_get_device_descriptor]
[ 3.034294] [0000050f] libusb: debug [libusb_get_device_descriptor]
[ 3.034310] [0000050f] libusb: debug [libusb_get_device_descriptor]
[ 3.034325] [0000050f] libusb: debug [libusb_get_device_descriptor]
[ 3.034341] [0000050f] libusb: debug [libusb_get_device_descriptor]
[ 3.034356] [0000050f] libusb: debug [libusb_get_device_descriptor]
[ 3.034371] [0000050f] libusb: debug [libusb_get_device_descriptor]
[ 3.034386] [0000050f] libusb: debug [libusb_get_device_descriptor]
[ 3.034474] [0000050f] libusb: debug [libusb_open] open 20.6
[ 3.034556] [0000050f] libusb: debug [usbi_add_pollfd] add fd 19 events 1
[ 3.034562] [0000050f] libusb: debug [darwin_open] device open for access
[ 3.034581] [00003a0b] libusb: debug [handle_events] poll() returned 1
[ 3.034596] [00003a0b] libusb: debug [handle_events] caught a fish on the control pipe
[ 3.034607] [00003a0b] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
[ 3.034611] [00003a0b] libusb: debug [libusb_try_lock_events] someone else is modifying poll fds
[ 3.034614] [00003a0b] libusb: debug [libusb_handle_events_timeout_completed] event handler was active but went away, retrying
[ 3.034618] [00003a0b] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 3.034621] [00003a0b] libusb: debug [handle_events] poll fds modified, reallocating
[ 3.034626] [00003a0b] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
node(25029,0x7fff7b097000) malloc: *** error for object 0x10180aac0: pointer being freed was not allocated
*** set a breakpoint in malloc_error_break to debug
[1]    25029 abort      node app.js
```

brandonros, 2015-11-12:
Actually, it seems .open() succeeds and that the problem is on Node's end cleaning up.

```
var usb = require('usb');

usb.setDebugLevel(4);

var device = usb.findByIds(0x0eb8, 0xf000);

device.open();

console.log('ok');

device.close();
```

```
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 3.056504] [0000050f] libusb: debug [libusb_get_device_list]
[ 3.056550] [0000050f] libusb: debug [discovered_devs_append] need to increase capacity
[ 3.056589] [0000050f] libusb: debug [libusb_get_device_descriptor]
[ 3.056632] [0000050f] libusb: debug [libusb_get_device_descriptor]
[ 3.056650] [0000050f] libusb: debug [libusb_get_device_descriptor]
[ 3.056666] [0000050f] libusb: debug [libusb_get_device_descriptor]
[ 3.056682] [0000050f] libusb: debug [libusb_get_device_descriptor]
[ 3.056697] [0000050f] libusb: debug [libusb_get_device_descriptor]
[ 3.056712] [0000050f] libusb: debug [libusb_get_device_descriptor]
[ 3.056727] [0000050f] libusb: debug [libusb_get_device_descriptor]
[ 3.056742] [0000050f] libusb: debug [libusb_get_device_descriptor]
[ 3.056757] [0000050f] libusb: debug [libusb_get_device_descriptor]
[ 3.056847] [0000050f] libusb: debug [libusb_open] open 20.6
[ 3.056925] [0000050f] libusb: debug [usbi_add_pollfd] add fd 19 events 1
[ 3.056931] [0000050f] libusb: debug [darwin_open] device open for access
[ 3.056954] [00003a0b] libusb: debug [handle_events] poll() returned 1
[ 3.056969] [00003a0b] libusb: debug [handle_events] caught a fish on the control pipe
[ 3.056979] [00003a0b] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
[ 3.056983] [00003a0b] libusb: debug [libusb_try_lock_events] someone else is modifying poll fds
[ 3.056987] [00003a0b] libusb: debug [libusb_event_handler_active] someone else is modifying poll fds
[ 3.056990] [00003a0b] libusb: debug [libusb_handle_events_timeout_completed] another thread is doing event handling
[ 3.057015] [00003a0b] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
[ 3.057020] [00003a0b] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 3.057023] [00003a0b] libusb: debug [handle_events] poll fds modified, reallocating
[ 3.057027] [00003a0b] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
ok
[ 3.061350] [0000050f] libusb: debug [libusb_close]
[ 3.061382] [00003a0b] libusb: debug [handle_events] poll() returned 1
[ 3.061387] [00003a0b] libusb: debug [handle_events] caught a fish on the control pipe
[ 3.061393] [00003a0b] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
[ 3.061397] [00003a0b] libusb: debug [libusb_try_lock_events] someone else is modifying poll fds
[ 3.061400] [00003a0b] libusb: debug [libusb_event_handler_active] someone else is modifying poll fds
[ 3.061403] [00003a0b] libusb: debug [libusb_handle_events_timeout_completed] another thread is doing event handling
[ 3.061423] [00003a0b] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
[ 3.061428] [00003a0b] libusb: debug [libusb_try_lock_events] someone else is modifying poll fds
[ 3.061432] [00003a0b] libusb: debug [libusb_event_handler_active] someone else is modifying poll fds
[ 3.061436] [00003a0b] libusb: debug [libusb_handle_events_timeout_completed] another thread is doing event handling
[ 3.061462] [0000050f] libusb: debug [usbi_remove_pollfd] remove fd 19
[ 3.061490] [00003a0b] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
[ 3.061500] [00003a0b] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 3.061506] [00003a0b] libusb: debug [handle_events] poll fds modified, reallocating
[ 3.061511] [00003a0b] libusb: debug [handle_events] poll() 2 fds with timeout in 60000ms
node(25077,0x7fff7b097000) malloc: *** error for object 0x101e009f0: pointer being freed was not allocated
*** set a breakpoint in malloc_error_break to debug
[1]    25077 abort      node app.js
```

kevinmehall, 2015-11-13:
Probably #100

asciidisco, 2015-11-16:
I´m observing exactly the same issue on Node 4.0.0 with the latest version from master.

resistdesign, 2015-11-22:
Same with Node 4

closed 2017-02-13 by kevinmehall