node-usb/node-usb #105: Sending a transfer to the device does nothing???

resistdesign, 2015-11-24:
Not sure if it's me or the library but I'm trying to get this device (Belkin f1u201 http://www.belkin.com/us/support-article?articleNum=4451) to switch by replicating the host-to-device transfer in this `pcap` file (https://drive.google.com/file/d/0ByjZaKpw22ObSEpvdWE3RTd0WTQ/view?usp=sharing). Nothing happens :(

Here is my code:
```
var usb = require('usb');

// Get the switch by its vendor and product ids.
var switchDevice = usb.findByIds(0x050D, 0x0201);

switchDevice.open();

switchDevice.controlTransfer(0x21, 9, 0x0200, 0, new Buffer(''), function (error, data) {
    if (error) {
        console.log('ERROR:', error);
    } else {
        console.log('SUCCESS:', data);
    }
});
```

Can someone tell me what I'm doing wrong?

kevinmehall, 2015-11-24:
The setup packet in the pcap file has a wLength of 2, while you're sending an empty buffer. The second packet in your pcap file looks like the OUT packet of the data phase, so try passing its data, `new Buffer([0x02, 0x00])`, as the data parameter.

closed 2015-11-24 by kevinmehall

resistdesign, 2015-11-24:
@kevinmehall HOLY F***ING SH**! That was it man! I hacked on that all weekend, what a dumb as I am :P Thank you!