node-usb/node-usb #108: Can't install on OSX 10.11.2 and node 5.1.1

infusion, 2016-01-07:
Hi, 

when trying to install node-usb on OSX, I get the following error:

```
npm ERR! argv "/usr/local/Cellar/node/5.1.1/bin/node" "/usr/local/bin/npm" "install" "usb"
npm ERR! node v5.1.1
npm ERR! npm  v3.5.2
npm ERR! path /Users/robert/node_modules/.staging/readable-stream-b78f7d3e
npm ERR! code ENOENT
npm ERR! errno -2
npm ERR! syscall rename

npm ERR! enoent ENOENT: no such file or directory, rename '/Users/robert/node_modules/.staging/readable-stream-b78f7d3e' -> '/Users/robert/node_modules/usb/node_modules/node-pre-gyp/node_modules/npmlog/node_modules/are-we-there-yet/node_modules/readable-stream'
npm ERR! enoent ENOENT: no such file or directory, rename '/Users/robert/node_modules/.staging/readable-stream-b78f7d3e' -> '/Users/robert/node_modules/usb/node_modules/node-pre-gyp/node_modules/npmlog/node_modules/are-we-there-yet/node_modules/readable-stream'
npm ERR! enoent This is most likely not a problem with npm itself
npm ERR! enoent and is related to npm not being able to find a file.
npm ERR! enoent 

npm ERR! Please include the following file with any support request:
npm ERR!     /Users/robert/npm-debug.log

```

taoyuan, 2016-01-09:
I have the same problem.

withgallantry, 2016-01-24:
I have this problem also

jrobeson, 2016-01-28:
the title of the bug should be updated, this doesn't work on linux either.

marcopiraccini, 2016-01-30:
+1. With node v4.2.0 and npm 3.6.0

jrobeson, 2016-01-30:
can you folks try with npm 2 and see if it works there?

marcopiraccini, 2016-01-30:
Yes, downgraded to npm 2 and it works.

jrobeson, 2016-01-30:
@infusion : if you can't replicate with npm 2, then the bug title should probably be changed.

infusion, 2016-01-31:
npm 2 works fine

charlesbrunelle, 2016-03-09:
It seems that the dependency to readable-streams is missing. Try
`npm install readable-stream`

Zozz, 2016-03-15:
Thanks! Installing readable-stream solved the problem for me.

jrobeson, 2018-08-17:
This doesn't occur in node 6.x does it?  5.x is EOL so I'd say this bug should be closed.

infusion, 2018-08-17:
With the newest setup it works for me. So I think it can be closed.

jrobeson, 2018-09-17:
I removed 5.x and 7.x from CI here https://github.com/tessel/node-usb/pull/253