node-usb/node-usb #109: [question] npm install fail on windows 10

j796160836, 2016-01-24:
Hi, I got some error during install `bleno` and I found that might be some issue about installing usb module.

```
lenovo-PC:bleno$ npm install usb

> usb@1.1.1 install C:\Users\lenovo\Documents\node\bleno\node_modules\usb
> node-pre-gyp install --fallback-to-build

Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  core.c
  descriptor.c
  hotplug.c
  io.c
C:\Users\lenovo\Documents\node\bleno\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: 'timespec':
'struct' type redefinition (compiling source file ..\libusb\libusb\descriptor.c) [C:\Users\lenovo\Documents\node\bleno\
node_modules\usb\build\libusb.vcxproj]
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: see declaration of 'timespec' (com
  piling source file ..\libusb\libusb\descriptor.c)
C:\Users\lenovo\Documents\node\bleno\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: 'timespec':
'struct' type redefinition (compiling source file ..\libusb\libusb\io.c) [C:\Users\lenovo\Documents\node\bleno\node_mod
ules\usb\build\libusb.vcxproj]
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: see declaration of 'timespec' (com
  piling source file ..\libusb\libusb\io.c)
..\libusb\libusb\io.c(1227): warning C4244: '=': conversion from 'time_t' to 'long', possible loss of data [C:\Users\le
novo\Documents\node\bleno\node_modules\usb\build\libusb.vcxproj]
C:\Users\lenovo\Documents\node\bleno\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: 'timespec':
'struct' type redefinition (compiling source file ..\libusb\libusb\hotplug.c)strerror.c [C:\Users\lenovo\Documents\node
\bleno\node_modules\usb\build\libusb.vcxproj]

  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: see declaration of 'timespec' (com
  piling source file ..\libusb\libusb\hotplug.c)..\libusb\libusb\io.c(1902): warning C4244: '=': conversion from 'time_
  t' to 'long', possible loss of data

..\libusb\libusb\io.c(2435): warning C4244: '=': conversion from 'time_t' to 'long', possible loss of data [C:\Users\le
novo\Documents\node\bleno\node_modules\usb\build\libusb.vcxproj]
C:\Users\lenovo\Documents\node\bleno\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: 'timespec':
'struct' type redefinition (compiling source file ..\libusb\libusb\core.c) [C:\Users\lenovo\Documents\node\bleno\node_m
odules\usb\build\libusb.vcxproj]
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: see declaration of 'timespec' (com
  piling source file ..\libusb\libusb\core.c)sync.c

  poll_windows.c
  threads_windows.c
C:\Users\lenovo\Documents\node\bleno\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: 'timespec':
'struct' type redefinition (compiling source file ..\libusb\libusb\strerror.c) [C:\Users\lenovo\Documents\node\bleno\no
de_modules\usb\build\libusb.vcxproj]
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: see declaration of 'timespec' (com
  piling source file ..\libusb\libusb\strerror.c)
  windows_usb.c
C:\Users\lenovo\Documents\node\bleno\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: 'timespec':
'struct' type redefinition (compiling source file ..\libusb\libusb\sync.c) [C:\Users\lenovo\Documents\node\bleno\node_m
odules\usb\build\libusb.vcxproj]
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: see declaration of 'timespec' (com
  piling source file ..\libusb\libusb\sync.c)
C:\Users\lenovo\Documents\node\bleno\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: 'timespec':
'struct' type redefinition (compiling source file ..\libusb\libusb\os\threads_windows.c)win_delay_load_hook.c [C:\Users
\lenovo\Documents\node\bleno\node_modules\usb\build\libusb.vcxproj]

  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: see declaration of 'timespec' (com
  piling source file ..\libusb\libusb\os\threads_windows.c)
..\libusb\libusb\os\threads_windows.c(194): warning C4244: '=': conversion from 'time_t' to 'long', possible loss of da
ta [C:\Users\lenovo\Documents\node\bleno\node_modules\usb\build\libusb.vcxproj]
..\libusb\libusb\os\threads_windows.c(196): warning C4244: '=': conversion from 'const time_t' to 'long', possible loss
 of data [C:\Users\lenovo\Documents\node\bleno\node_modules\usb\build\libusb.vcxproj]
C:\Users\lenovo\Documents\node\bleno\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: 'timespec':
'struct' type redefinition (compiling source file ..\libusb\libusb\os\poll_windows.c) [C:\Users\lenovo\Documents\node\b
leno\node_modules\usb\build\libusb.vcxproj]
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: see declaration of 'timespec' (com
  piling source file ..\libusb\libusb\os\poll_windows.c)
C:\Users\lenovo\Documents\node\bleno\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: 'timespec':
[... 49 lines cut ...]
npm ERR! usb@1.1.1 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the usb@1.1.1 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the usb package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls usb
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     C:\Users\lenovo\Documents\node\bleno\npm-debug.log
lenovo-PC:bleno$
```

Some info I had install
1. Python 2.7.11
2. Visual Studio Community 2015 Edition
3. node-gyp@3.2.1
4. node@5.3.0

Any help or suggestion is appreciate.

j796160836, 2016-01-24:
I found that might be compile error in `libusb` mobule.
I open another thread on here.
https://github.com/libusb/libusb/issues/144
Thanks.

kevinmehall, 2016-01-25:
`node-usb` brings its own copy of libusb, so any fix upstream will have to be brought in.

AlexDmr, 2016-02-21:
I experience the same problem, apparently the last version of libusb has taken the problem into account and it should be solved.
I don't know how to force node-usb to use git to retrieve libusb. Is there a way ?
Otherwise when will node-usb be updated ?

rwaldron, 2016-02-21:
This is the same error causing t2-cli builds on Travis-CI to fail

kevinmehall, 2016-02-21:
Libusb is a git submodule at the root of this repository, and you can update to libusb master like so:

```
git clone git@github.com:nonolith/node-usb.git --recursive
cd node-usb/libusb
git remote add upstream https://github.com/libusb/libusb.git
git fetch upstream
git checkout upstream/master
cd ..
npm install
```

The complication is that this library using a branch of libusb with a patch to [support hotplug events on Windows](https://github.com/libusb/libusb/issues/86), so using upstream libusb will break that functionality. The [windows hotplug branch](https://github.com/kevinmehall/libusb/tree/hp) has to be rebased again.

AlexDmr, 2016-02-22:
Thanks for this procedure, it will be very usefull to test things in a first time. 
Any idea when the window hotplug branch will be updated (I mean is it a mater of weeks / months) ?

mjohnst, 2016-02-23:
I'm getting an additional error when checking out the newer upstream master for libusb:
`c1 : fatal error C1083: Cannot open source file: '..\libusb\libusb\os\windows_usb.c': No such file or directory [C:\Users\mjohnst\AppData\Roaming\npm\node_modules\usb\build\libusb.vcxproj]`

It seams like `windows_usb.c` and `windows_usb.h` were replaced with `windows_usbdk.c`, `windows_usbdk.h`, `windows_winusb.c`, and `windows_winusb.h`.
Any ideas on how I can fix this further?

kevinmehall, 2016-02-24:
@mjohnst node-usb builds libusb with `gyp` instead of the makefiles included with libusb, so you have to update the file names in `libusb.gypi`. The `winusb` files correspond to the backend that was used previously.

trmck, 2016-02-24:
These are the errors I receive after updating libusb.gypi to include the renamed files in libusb

..\src\device.cc(122): error C2332: 'struct': missing tag name [c:\users\trmck\npm_modules\node-usb\build\usb_bindings.vcxproj]
..\src\device.cc(122): error C2039: '<unnamed-tag>': is not a member of 'libusb_config_descriptor' [c:\users\trmck\npm_modules\node-usb\build\usb_bindings.vcxproj]
  c:\users\trmck\npm_modules\node-usb\libusb\libusb\libusb.h(643): note: see declaration of 'libusb_config_descriptor'
..\src\device.cc(122): error C2059: syntax error: '[' [c:\users\trmck\npm_modules\node-usb\build\usb_bindings.vcxproj]
..\src\device.cc(124): error C2146: syntax error: missing ';' before identifier 'v8altsettings' [c:\users\trmck\npm_modules\node-usb\build\usb_bindings.vcxproj]
..\src\device.cc(124): error C2275: 'v8::Local<v8::Array>': illegal use of this type as an expression [c:\users\trmck\npm_modules\node-usb\build\usb_bindings.vcxproj]
..\src\device.cc(124): error C2039: 'v8altsettings': is not a member of 'libusb_config_descriptor' [c:\users\trmck\npm_modules\node-usb\build\usb_bindings.vcxproj]
  c:\users\trmck\npm_modules\node-usb\libusb\libusb\libusb.h(643): note: see declaration of 'libusb_config_descriptor'
..\src\device.cc(124): error C2039: 'numAltSettings': is not a member of 'libusb_config_descriptor' [c:\users\trmck\npm_modules\node-usb\build\usb_bindings.vcxproj]
  c:\users\trmck\npm_modules\node-usb\libusb\libusb\libusb.h(643): note: see declaration of 'libusb_config_descriptor'
..\src\device.cc(125): error C2039: 'v8interfaces': is not a member of 'libusb_config_descriptor' [c:\users\trmck\npm_modules\node-usb\build\usb_bindings.vcxproj]
  c:\users\trmck\npm_modules\node-usb\libusb\libusb\libusb.h(643): note: see declaration of 'libusb_config_descriptor'
..\src\device.cc(125): error C2039: 'Set': is not a member of 'libusb_config_descriptor' [c:\users\trmck\npm_modules\node-usb\build\usb_bindings.vcxproj]
  c:\users\trmck\npm_modules\node-usb\libusb\libusb\libusb.h(643): note: see declaration of 'libusb_config_descriptor'
..\src\device.cc(125): error C2065: 'v8altsettings': undeclared identifier [c:\users\trmck\npm_modules\node-usb\build\usb_bindings.vcxproj]
..\src\device.cc(129): error C2332: 'struct': missing tag name [c:\users\trmck\npm_modules\node-usb\build\usb_bindings.vcxproj]
..\src\device.cc(129): error C2039: '<unnamed-tag>': is not a member of 'libusb_config_descriptor' [c:\users\trmck\npm_modules\node-usb\build\usb_bindings.vcxproj]
  c:\users\trmck\npm_modules\node-usb\libusb\libusb\libusb.h(643): note: see declaration of 'libusb_config_descriptor'
..\src\device.cc(129): error C2059: syntax error: '[' [c:\users\trmck\npm_modules\node-usb\build\usb_bindings.vcxproj]
..\src\device.cc(131): error C2146: syntax error: missing ';' before identifier 'v8idesc' [c:\users\trmck\npm_modules\node-usb\build\usb_bindings.vcxproj]
..\src\device.cc(131): error C2275: 'v8::Local<v8::Object>': illegal use of this type as an expression [c:\users\trmck\npm_modules\node-usb\build\usb_bindings.vcxproj]
..\src\device.cc(131): error C2039: 'v8idesc': is not a member of 'libusb_config_descriptor' [c:\users\trmck\npm_modules\node-usb\build\usb_bindings.vcxproj]
  c:\users\trmck\npm_modules\node-usb\libusb\libusb\libusb.h(643): note: see declaration of 'libusb_config_descriptor'
..\src\device.cc(132): error C2039: 'v8altsettings': is not a member of 'libusb_config_descriptor' [c:\users\trmck\npm_modules\node-usb\build\usb_bindings.vcxproj]
  c:\users\trmck\npm_modules\node-usb\libusb\libusb\libusb.h(643): note: see declaration of 'libusb_config_descriptor'
..\src\device.cc(132): error C2039: 'Set': is not a member of 'libusb_config_descriptor' [c:\users\trmck\npm_modules\node-usb\build\usb_bindings.vcxproj]
  c:\users\trmck\npm_modules\node-usb\libusb\libusb\libusb.h(643): note: see declaration of 'libusb_config_descriptor'
..\src\device.cc(132): error C2065: 'v8idesc': undeclared identifier [c:\users\trmck\npm_modules\node-usb\build\usb_bindings.vcxproj]
..\src\device.cc(134): error C2065: 'v8idesc': undeclared identifier [c:\users\trmck\npm_modules\node-usb\build\usb_bindings.vcxproj]
..\src\device.cc(134): error C2227: left of '->ForceSet' must point to class/struct/union/generic type [c:\users\trmck\npm_modules\node-usb\build\usb_bindings.vcxproj]
  ..\src\device.cc(134): note: type is 'unknown-type'
..\src\device.cc(135): error C2065: 'v8idesc': undeclared identifier [c:\users\trmck\npm_modules\node-usb\build\usb_bindings.vcxproj]
..\src\device.cc(135): error C2227: left of '->ForceSet' must point to class/struct/union/generic type [c:\users\trmck\npm_modules\node-usb\build\usb_bindings.vcxproj]
  ..\src\device.cc(135): note: type is 'unknown-type'
..\src\transfer.cc(68): warning C4267: '=': conversion from 'size_t' to 'int', possible loss of data [c:\users\trmck\npm_modules\node-usb\build\usb_bindings.vcxproj]
..\src\device.cc(136): error C2065: 'v8idesc': undeclared identifier [c:\users\trmck\npm_modules\node-usb\build\usb_bindings.vcxproj]
..\src\device.cc(136): error C2227: left of '->ForceSet' must point to class/struct/union/generic type [c:\users\trmck\npm_modules\node-usb\build\usb_bindings.vcxproj]
  ..\src\device.cc(136): note: type is 'unknown-type'
..\src\device.cc(137): error C2065: 'v8idesc': undeclared identifier [c:\users\trmck\npm_modules\node-usb\build\usb_bindings.vcxproj]
[... cut ...]
node-pre-gyp ERR! command "c:\\Program Files\\NodejsUwp\\Console\\node.exe" "c:\\Users\\trmck\\npm_modules\\node-usb\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd c:\users\trmck\npm_modules\node-usb
node-pre-gyp ERR! node -v v5.5.0-0
node-pre-gyp ERR! node-pre-gyp -v v0.6.21
node-pre-gyp ERR! not ok
Failed to execute 'node-gyp.cmd build --fallback-to-build --module=c:\users\trmck\npm_modules\node-usb\src\binding\usb_bindings.node --module_name=usb_bindings --module_path=c:\users\trmck\npm_modules\node-usb\src\binding' (1)

npm ERR! Windows_NT 10.0.10586

timfish, 2016-03-09:
Until this has been fixed, what is the latest version of `node-usb` which still compiles on Windows 10?

kevinmehall, 2016-03-09:
No version is known to compile on Windows 10 / Visual Studio 2015.

As far as I know, this is only a build issue -- binaries built on older versions of Windows and/or Visual Studio will work. This includes the binaries you should be getting automatically from node-pre-gyp. (If node-pre-gyp isn't working, that's a separate issue)

The next version of libusb is supposed to include Windows hotplug support. I do not plan to take further action on this until that is released, and I can drop that version into the build. In the mean time, if you want to **use** the library on Windows 10, use binaries from node-pre-gyp. If you want to build the library from source, either use an older version of Windows/Visual Studio, use upstream libusb and lose hotplug support, or help libusb get hotplug support upstream.

I'm no longer using Node or this library personally or professionally, so the amount of time I can put into maintaining this library is very limited.

timfish, 2016-03-09:
@kevinmehall No worries, thanks for taking the time to reply. 

I'm attempting to use a library which uses `node-usb` in [Electron](http://electron.atom.io/) shell. To use native libraries in Electron they have to be compiled against the customised version of node that they're using. This normally works seamlessly using gyp but is an issue here. If this becomes a problem for me I'll attempt it again with VS2012/13 and/or Windows 7.

Many thanks.

underscorebrody, 2016-03-11:
@timfish I'm working on the same thing, so far my best approach has been to bump down to `node` version 4.4.0 (which is the latest where NODE_MODULE_VERSION is 46) in order to use the latest precompiled binaries for `node-usb` (which are built for v46). To allow for this, I've also bumped down to `electron-prebuilt` version 0.35.5 which is the latest that still used `node` 4.4.0 internally. I'm still working on getting everything functional, hitting some ia32 vs. x64 architecture issues. Anyway, hope this helps until we can get a proper build in 47

timfish, 2016-03-12:
@underscorebrody Thanks, that's what I was looking for!

underscorebrody, 2016-03-13:
@timfish i ended up having to rebuild the module since the prebuilt didn't like running in virtualbox, i was able to do this by following these steps: 
- install visual studio 2013 (https://go.microsoft.com/fwlink/?LinkId=532495&clcid=0x409)
- `npm install usb` to get the prebuilt binary
- `cd node_modules/usb`
- Modify `binding.gyp` and add the following to `"variables"`:
    - `'module_name': 'usb_bindings', 'module_path': './src/binding'`
- `node-gyp rebuild --target=0.35.5 --arch=ia32 --dist-url=https://atom.io/download/atom-shell --msvs_version=2013`

The thread at https://discuss.atom.io/t/use-node-usb-inside-electron/16748 was a big help in figuring this out.

yangpu, 2016-08-17:
Is there any "stable version" to be installed in Windows 10 and Ubuntu 14.04? I can't install latest node-usb on both OS.

machinekoder, 2016-09-15:
@yangpu on Ubuntu you may need `libusb-dev` and `libudev-dev`