node-usb/node-usb #111: Does node-usb support bi-directional endpoints?

adjavaherian, 2016-02-04:
I'm trying to write a bulk transfer for the [crazy radio dongle](https://wiki.bitcraze.io/doc:crazyradio:usb:index) and they say that their endpoint 1 supports in and out, which seems to be valid in chrome.usb's bulkTransfer mechanism.  Does node-usb take this directional argument?  Perhaps in the data packet?  I know you say that isosynchronous is not supported, but this is different, right?  It looks like chrome.usb supports endpoint direction in its bulkTransfer and isosync is totally different.

In the following tests, you should be able to see that the endpoint is labeled with out, but the first test succeeds in reading.  The second test fails because the endpoint "in" needs to be read so the buffer can be cleared, however, I'm not sure how to do that in this scenario.  Any ideas, or is this even supported?

Thanks!

```javascript
describe.only('Packet tests', function() {

        var iface = null;

        beforeEach(function() {
            radio = new Crazyradio();
            radio.debugLevel(4);
            device = radio.find();
        });

        afterEach(function(done){
            iface.release(true, function(err){
                console.log('releasing', err);
                radio.close(device, function(err, data){
                    console.log('after', err);
                    done();
                });
            });
        });

        it('should read a packet', function(done){
            radio.open(device, function(){

                iface = device.interface(0);
                iface.claim();

                var radioIn = iface.endpoints[1];

                console.log('radioIn', radioIn.direction, radioIn.transferType, radioIn.descriptor);

                radioIn.transfer(64, function(err, data){
                    console.log('in err / data', err, data);
                    done();
                });
            });
        });

        it('should send a packet', function(){

            radio.open(device, function(){

                iface = device.interface(0);
                iface.claim();

                var radioOut = iface.endpoints[1];

                console.log('radioOut', radioOut.direction, radioOut.transferType, radioOut.descriptor);

                //Preparing commander packet
                var packet = new ArrayBuffer(15);
                var dv = new DataView(packet);

                dv.setUint8(0, 0x30, true);      // CRTP header
                dv.setFloat32(1, 0.0, true);    // Roll
                dv.setFloat32(5, 0.0, true);   // Pitch
                dv.setFloat32(9, 0.0, true);     // Yaw
                dv.setUint16(13, 0.0, true);  // Thrust

                radioOut.transfer(packet, function(error){
                    console.log('out transfer callback', error);
                    done();
                });
            });
[... 98 lines cut ...]
[ 0.106248] [00000513] libusb: debug [ep_to_pipeRef] pipe 2 on interface 0 matches
      ✓ should send a packet 
[ 0.107096] [00004503] libusb: debug [libusb_release_interface] interface 0
releasing undefined
      1) "after each" hook

  2 passing (22ms)
  1 failing

  1) CrazyRadio tests Packet tests "after each" hook:
     Uncaught Error: Can't close device with a pending request
      at Error (native)
      at Device.usb.Device.close (/Users/4m1r/Desktop/Crazyflie/crazyradio-node-driver/node_modules/usb/usb.js:43:7)
      at Crazyradio.close (/Users/4m1r/Desktop/Crazyflie/crazyradio-node-driver/index.js:61:12)
      at Interface.<anonymous> (/Users/4m1r/Desktop/Crazyflie/crazyradio-node-driver/test.js:463:23)
      at Device.<anonymous> (/Users/4m1r/Desktop/Crazyflie/crazyradio-node-driver/node_modules/usb/usb.js:200:7)

npm ERR! Test failed.  See above for more details.
[4m1r@Korgi:14:12:28 ~/Desktop/Crazyflie/crazyradio-node-driver]
$ 

```

Here's the driver and the rest of the tests, which work fine for configuration
https://github.com/adjavaherian/crazyradio-node-driver

kevinmehall, 2016-02-04:
USB endpoints aren't bidirectional, but the MSB of the endpoint address specifies the direction, and sometimes that bit is omitted when the direction is also specified. So there are actually two endpoints, `0x1` OUT and `0x81` IN, which `node-usb` will see as two separate endpoints.

closed 2016-02-04 by kevinmehall

adjavaherian, 2016-02-05:
Thanks @kevinmehall, that's what I'm seeing, but for some reason, my read from `0x81` never returns the callback.  Any ideas?

Here's the spec

```javascript
before(function() {
            radio = new Crazyradio();
            radio.debugLevel(4);
            device = radio.find();
        });

        afterEach(function(done){
            var iface = device.interfaces[0];
            iface.release(true, function(err){ console.log(err) });
            done();
        });

        it.only('should read a packet', function(done){

            console.log('opening', device);
            device.open();
            console.log('open', device.interfaces[0]);

            var iface = device.interfaces[0];
                iface.claim();

            console.log('endpoints[0]', iface.endpoints[0]);

            var radioIn = iface.endpoints[0];
            var radioOut = iface.endpoints[1];

            console.log('radioOut', radioOut.direction, radioOut.transferType, radioOut.descriptor.bEndpointAddress);
            console.log('radioIn', radioIn.direction, radioIn.transferType, radioIn.descriptor.bEndpointAddress);

            radioOut.transfer([1], function(err){
                console.log('out cb', err);

                assert.equal(radioIn, iface.endpoint(0x81));

                radioIn.transfer(64, function(err, data){
                    console.log('in err / data', err, data);
                    done();
                });
            });
        });
```

and the log

```
> crazyradio-node-driver@0.0.1 test /Users/4m1r/Desktop/Crazyflie/crazyradio-node-driver
> mocha test.js

  CrazyRadio tests
    Packet tests
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.023562] [00000513] libusb: debug [libusb_get_device_list] 
[ 0.023735] [00000513] libusb: debug [libusb_get_device_descriptor] 
[ 0.023856] [00000513] libusb: debug [libusb_get_device_descriptor] 
[ 0.023924] [00000513] libusb: debug [libusb_get_device_descriptor] 
[ 0.024010] [00000513] libusb: debug [libusb_get_device_descriptor] 
[ 0.024067] [00000513] libusb: debug [libusb_get_device_descriptor] 
opening { busNumber: 20,
  deviceAddress: 26,
  deviceDescriptor: 
   { bLength: 18,
     bDescriptorType: 1,
     bcdUSB: 512,
     bDeviceClass: 0,
[... 128 lines cut ...]
[ 0.034513] [00000513] libusb: debug [ep_to_pipeRef] pipe 2 on interface 0 matches
[ 0.034681] [0000230f] libusb: debug [darwin_async_io_callback] an async io operation has completed
[ 0.034715] [00003f03] libusb: debug [handle_events] poll() returned 1
[ 0.034725] [00003f03] libusb: debug [op_handle_events] checking fd 22 with revents = 1
[ 0.034747] [00003f03] libusb: debug [darwin_handle_callback] handling bulk completion with kernel status 0
[ 0.034756] [00003f03] libusb: debug [usbi_handle_transfer_completion] transfer 0x1039007a8 has callback 0x103b51935
[ 0.034775] [00003f03] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
[ 0.034781] [00003f03] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.034786] [00003f03] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
out cb undefined
[ 0.035264] [00000513] libusb: debug [ep_to_pipeRef] converting ep address 0x81 to pipeRef and interface
[ 0.035291] [00000513] libusb: debug [ep_to_pipeRef] pipe 1 on interface 0 matches
      1) should read a packet
[ 2.034030] [00004703] libusb: debug [libusb_release_interface] interface 0
undefined

  0 passing (2s)
  1 failing

  1) CrazyRadio tests Packet tests should read a packet:
     Error: timeout of 2000ms exceeded. Ensure the done() callback is being called in this test.
      at null.<anonymous> (/usr/local/lib/node_modules/mocha/lib/runnable.js:170:19)
      at Timer.listOnTimeout (timers.js:119:15)

```

kevinmehall, 2016-02-06:
That looks right, so I'd guess that the device is simply not sending you a packet within the timeout. You might have to send it a control transfer to enable something, or a command that makes it reply.