node-usb/node-usb #114: Isochronous transfers?

jandrieu, 2016-02-21:
The readme makes it clear node-usb doesn't support isochronous transfers.

What would it take to make that work?

kevinmehall, 2016-02-21:
Isochronous transfers have a slightly different [API](http://libusb.sourceforge.net/api-1.0/structlibusb__iso__packet__descriptor.html) where the packets are handled individually. The C++ Transfer class would have to be extended to [allocate](https://github.com/nonolith/node-usb/blob/760ad3638a4673b32e6cedb2e7b592b7547511a7/src/transfer.cc#L12) transfers with a specified number of isochronous packet descriptors and expose their fields to JS.

cinderblock, 2019-08-05:
How much work would this actually take to implement? I'm considering giving this a shot but I'm not super versed in low level USB stuff. Is there anyone else that's willing to take a look?

Apaev, 2020-03-13:
Will Isochronous transfers be implemented?

cinderblock, 2020-03-13:
I wonder if this would lend itself more to a rewrite. Might also be able to expose what I did in #326 as a full featured high performance API.

mcuee, 2023-05-06:
From https://github.com/node-usb/node-usb/issues/452#issuecomment-945181798
> I have no plans to implement this soon. A google search will uncover how libusb can be used to do these transfers, but I get the impression it's challenging. Good luck!

I agree isochronous transfer is tough to get right.

As of now, people are still reporting issues for libusb isochronous transfer, especially on Windows and Android.
https://github.com/libusb/libusb/issues?q=is%3Aopen+is%3Aissue+label%3Aisoc

mcuee, 2023-05-06:
For those who want to work on this feature, one of the first hurdle is to find the right known-good isoc firmware.I will recommend to get a Cypress EZUSB-FX3 demo board or EZUSB FX2LP as Cypress (now part of Infineon) provides known-good firmware like iso_loop and isp_source_sink.

* https://github.com/libusb/libusb/wiki/FAQ#user-content-Do_you_have_a_list_of_known_good_devices_to_test_libusb