node-usb/node-usb #122: Determine the (Windows) path of an attached device

Jordyderuijter, 2016-05-11:
Hi there!

How can I determine the path of an attached device in order to read a file from a USB device?

I'm running node on Windows and I want to read a specific XML file from a USB device when it's being attached.

Thanks in advance!

kevinmehall, 2016-05-11:
I'm assuming you're talking about a USB Mass Storage Class device, like a flash drive or hard drive. In that case, this library isn't what you're looking for.

This is for low-level raw access to USB endpoints. In the case of a mass storage device, you're using the driver in the kernel instead of writing a driver in Node with this library.

closed 2016-05-11 by kevinmehall

Jordyderuijter, 2016-05-11:
Thanks for your answer!

I'm quite a newbie in the low-level world. I was indeed talking about a USB flash drive device. I want to be able to put an XML configuration file on a USB drive from any pc. This configuration file should be loaded automatically by Node when inserted. 

Does this mean there is no way to accomplish this in Node without writing any external driver / script?