node-usb/node-usb #124: Crash with node-usb sending to sleep

armaseg, 2016-05-30:
- Operating system: Windows 7 64bits
- Node: 6.1.0

I've been using node-usb for a while and It works really good but it crashs only when I set the pc on sleep mode.

You can reproduce this behavior using this code:

````javascript
var usb = require('usb');
usb.on('attach', function(device) { console.log('usb connected') });
````
and then set the PC on sleep mode.

Regards,

gfcittolin, 2016-06-05:
I confirm the behavior. Actually, for me, node just exits without any message, but with the exit code `-1073741819`

Using Windows 8.1 and Node 4.2.1

jeanfsal, 2016-06-06:
I have the same issue on windows 10....anybody knows how to solve this.

I am ready to pay for the fix...

taktran, 2016-07-26:
We are seeing the same issue too, when [noble](https://github.com/sandeepmistry/noble) is used on windows with a dongle.

rstolwijk, 2016-10-18:
We are also experiencing this issue on Windows 7 and Windows 8.1 using nw.js v0.15.4
and Node v6.2.2

pablozandona, 2016-12-08:
I have the same issue on:

- windows 7
- node 6.3.1
- nwjs 0.16.1

bladesling, 2018-08-10:
@stberger3040 that does indeed fix this issue. Running this in nwjs and no more crashing on sleep. Thank you!

stberger3040, 2021-10-15:
Thank you @thegecko for taking care of this, I really don't have time. Beware, however that ist is really a quickfix. Devices with an interface path longer than 4096 will stop working with that library. I assumed that this is never the case here (which could be a wrong assumption). The action problem to cause the crash was an integer underflow. `len` was `-1` (which leads to a quite high number when interpreted as `unsigned` ;-)

thegecko, 2021-10-15:
Thanks @stberger3040