node-usb/node-usb #125: Failed at the usb@1.1.2 install script 'node-pre-gyp install --fallback-to-build'.

phillip-hogan, 2016-06-04:
Hi,

I keep getting this issue above when I try to npm install. I have the latest version of node and npm and am running this on a raspberry pi. See terminal output below. Any help would be great - thanks:

Failed to execute '/usr/local/bin/node /home/pi/arm/owi-robot-arm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/pi/arm/owi-robot-arm/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/home/pi/arm/owi-robot-arm/node_modules/usb/src/binding' (1)
npm ERR! Linux 3.18.11+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install"
npm ERR! node v6.2.1
npm ERR! npm  v3.9.5
npm ERR! code ELIFECYCLE

npm ERR! usb@1.1.2 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the usb@1.1.2 install script 'node-pre-gyp install --fallback-to-build'.

Liriel, 2016-06-09:
Same here on Windows 8.1 64

>error C2011: 'timespec': 'struct' type redefinition (compiling source file ..\libusb\libusb\io.c
this seems to be a problem with libusb/node-usb https://github.com/nonolith/node-usb/issues/109

vpetroff, 2016-06-16:
Same here on Arch Linux on a Pi 3

nimblez, 2016-06-28:
I ran into the same problem. Anyone having better luck with this?

mbifulco, 2016-07-26:
Same here on Raspbian on a Pi 3, Windows 10, OS X Yosemite, and Bash on Ubuntu on Windows.  Help!

anne-decusatis, 2016-07-30:
We saw this on Ubuntu 12.04 and were able to fix by installing g++ 4.8 similar to here https://askubuntu.com/questions/26498/choose-gcc-and-g-version (the default g++ on Ubuntu 12.04 is 4.6, which doesn't support c++11 extensions used in this?)

Roarster31, 2016-08-02:
On Mac OS, I tried using [c++11](http://stackoverflow.com/questions/18139710/using-c11-in-macos-x-and-compiled-boost-libraries-conundrum) but my problem wasn't solved until I fully reinstalled Node and Npm using [this gist](https://gist.github.com/DanHerbert/9520689). I was using the latest Node and Npm versions beforehand (Node 6.3.1 & Npm 3.10.5) so it may have been the removal of all global node modules that did the trick. I'm not sure...

ravikp7, 2017-03-01:
I'm also having the same issue. I've included the log file. Somebody please take a look and help.
[npm-debug.log.zip](https://github.com/tessel/node-usb/files/811038/npm-debug.log.zip)

ravikp7, 2017-03-01:
I installed the package libudev-dev and then tried to install usb and the problem was fixed

woodev01, 2017-03-09:
What is this issue? Honestly? I tried this module on both windows and Angstrom and I've gotten the same error both times. I have libudev installed.

error Linux 4.1.15-v2.6b2+g0ff849d
error argv "/home/root/node-v7.7.2-linux-armv7l/bin/node" "/home/root/node-v7.7.2-linux-armv7l/bin/npm" "install" "usb"
error node v7.7.2
error npm  v4.1.2
error code ELIFECYCLE
error usb@1.2.0 install: `node-pre-gyp install --fallback-to-build`
error Exit status 1
error Failed at the usb@1.2.0 install script 'node-pre-gyp install --fallback-to-build'.

Does anyone know how to fix this recurring error?

kevinmehall, 2017-03-09:
@woodev01 This and the other logs posted in this thread are NPM errors saying the build command failed, but without any details what went wrong. For this to be actionable, please post the full log including the error messages from the compiler.

woodev01, 2017-03-21:
@kevinmehall - Here is my log. It's stopped doing `node-pre-gyp install --fallback-to-build` but is still having serious errors.

[npm-debug.TXT](https://github.com/tessel/node-usb/files/858722/npm-debug.TXT)

jrobeson, 2019-03-19:
node-pre-gyp is no longer used as of https://github.com/tessel/node-usb/pull/300 , so this can be closed too.

closed 2019-03-19 by kevinmehall