node-usb/node-usb #128: Error: self signed certificate in certificate chain while installing from proxy

panneerselvampst, 2016-06-22:
gyp WARN install got an error, rolling back install
gyp ERR! configure error
gyp ERR! stack Error: self signed certificate in certificate chain
gyp ERR! stack     at Error (native)
gyp ERR! stack     at TLSSocket.<anonymous> (_tls_wrap.js:1017:38)
gyp ERR! stack     at emitNone (events.js:67:13)
gyp ERR! stack     at TLSSocket.emit (events.js:166:7)
gyp ERR! stack     at TLSSocket._init.ssl.onclienthello.ssl.oncertcb.TLSSocket._finishInit (_tls_wrap.js:582:8)
gyp ERR! stack     at TLSWrap.ssl.onclienthello.ssl.oncertcb.ssl.onnewsession.ssl.onhandshakedone (_tls_wrap.js:424:38)
gyp ERR! System Windows_NT 6.1.7601

kevinmehall, 2016-06-22:
This is not an issue in node-usb. See https://github.com/nodejs/node-gyp/issues/695 and please Google the error message before reporting an issue.

closed 2016-06-22 by kevinmehall

panneerselvampst, 2016-06-23:
Thanks for your response.