node-usb/node-usb #129: DLL error when importing module on Windows 10.

amlinger, 2016-06-23:
I am trying to use node-usb in an Electron app, that is planned to run on multiple platforms, mainly OSX and Windows. On OSX, the app is running hassle-free, but with windows the following message is shown when starting the app up, and importing the node-usb library.

```
App threw an error when running Error: A dynamic link library (DLL) initialization routine failed.
\\?\C:\Users\me\electron-app\app\node_modules\usb\src\binding\usb_bindings.node
    at Error (native)
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:158:20)
    at Object.Module._extensions..node (module.js:568:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:158:20)
    at Module.load (module.js:456:32)
    at tryModuleLoad (module.js:415:12)
    at Function.Module._load (module.js:407:3)
    at Module.require (module.js:466:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (C:\Users\me\electron-app\app\node_modules\usb\usb.js:5:38)
```

The app is running Electron 1.2.5, which in it's turn runs node 6.1.0, as do I locally, on Windows 10.

Does anyone have any ideas?

stberger3040, 2016-08-02:
I've got the same problem. My first guess is that the binary does not work with such high node versions. Will have a further look at the issue soon...

stberger3040, 2016-08-03:
Ok, I use electron 1.3.1 which uses v8 version 5.2.361.43. It seems nodejs 6x use v8 5.0.x (at least on windows). This is the problem...one possible solution would be to compile the module with v8 5.2.x dependency. Not sure how to accomplish that

stberger3040, 2016-08-03:
The solution can be found [here](https://github.com/electron/electron/blob/master/docs/tutorial/using-native-node-modules.md)
As mentioned before you need to build the usb module with the same v8 version as electron uses

tcr, 2016-09-02:
Thanks @stberger3040.

That's the correct answer, you'll have to use --build-from-source @amlinger.

closed 2016-09-02 by tcr