node-usb/node-usb #134: use webpack error

qqnp1100, 2016-08-15:
node－usb in my project is work,but when i run webpack ,it's log
` Cannot find module "./abi_crosswalk.json"`
than i add json-loader in webpack 
`{ test: /\.json$/, loader: 'json-loader' },`
run ,it's log:
`package.json does not exist at /package.json`
How can I deal with it，thanks

rwaldron, 2016-08-15:
You won't be able to webpack node-usb, it's a compiled add-on (not pure JS)

qqnp1100, 2016-08-16:
@rwaldron 
Is there someway to use node-usb in webpack project?

reconbot, 2016-08-16:
It will only work with nodejs, can't run in the browser directly.

qqnp1100, 2016-08-16:
@reconbot 
I know,My project development on electron,it's have nodejs. 
without webpack , electron app use node-usb is ok.
I want use es6 ,so should use webpack. 
I don't know how to solve it @

reconbot, 2016-08-16:
You can use babel without webpack I know that works well. (babel-register works
great for example). It would be a bwebpack issue to make it work with
compiled modules however. They probably don't support it out of the box as
it's not a usual kind of require. I did find a thread saying you can have
webpack ignore your node modules. (
https://github.com/webpack/webpack/issues/839) that might solve the problem
too.

-Francis

---
Francis Gulotta
wizard@roborooter.com

qqnp1100, 2016-08-16:
@reconbot thank you vary much,I find way use babel without webpack,now it's ok .
Thanks again (｡◕ˇ∀ˇ◕）

closed 2016-08-16 by qqnp1100