node-usb/node-usb #139: Issue installing on raspberry PI 3

rob-cowsley, 2016-09-16:
Having some trouble installing using npm on the Raspberry PI 3 as per the attached screenshot and log. 

Error: `make` failed with exit code: 2

julien-digilink, 2016-09-28:
Same issue for me on m'y Rasperry Pi 3

rob-cowsley, 2016-09-28:
I ended up switching to a different distro to get this working.

closed 2016-09-28 by rob-cowsley

julien-digilink, 2016-10-04:
@rob-cowsley which one please ? I still experiment the same problem. Thanks

rob-cowsley, 2016-10-04:
I used he pre-built kali image as a base from here:

http://docs.kali.org/kali-on-arm/install-kali-linux-arm-raspberry-pi

georgkreimer, 2017-11-22:
Really? How can this be the solution?

pr1ntr, 2018-04-10:
seriously?

rwaldron, 2018-04-10:
@pr1ntr @georgkreimer Looks like this was a permissions problem—how would you suggest that ths project fix permissions on a device or machine that it has no access to?

pr1ntr, 2018-04-10:
ok so apparently this was an issue with 1.2. 1.3.1 it works fine. I was trying to use it with 
`node-fadecandy` and their npm package used 1.2. So I was able to install just fine on a raspberry pi without sudo. Sorry for the snide remark.

rwaldron, 2018-04-12:
I'm glad everything is working now ❤️