node-usb/node-usb #140: Error in Electron

woodev01, 2016-09-19:
I'm running Windows 7 in 64 bit and ran into a weird error during compilation of my Electron App. As it was attempting to launch I got this error (all the app is doing with node-usb is requiring it).

A Javascript error occurred in the main process:

Uncaught Exception:
Error: The system cannot find message text for message number 0x%1 in the message file for %2.
\\?\C:\Users\{{path}}\usb_bindings.node
  at Error (native) ...

I have no idea what this means. The only similar issue I could find was this issue 
>> https://github.com/brave/browser-laptop/issues/350

They suggested that it could be an architecture interference issue, because I'm using 64-bit. Any ideas how to move on with this??

rwaldron, 2016-09-19:
Did you rebuild node-usb for Electron?

woodev01, 2016-09-20:
I hadn't at the time of posting this issue. I now have and I am still getting this same error.

rwaldron, 2016-09-20:
Ok, thanks for the follow up. Can you create a test case for this? If you put it into a repo, then I can clone it and work on reproduction locally (on windows machine). Without that, I'd have to try to guess to piece it together

woodev01, 2016-09-20:
Literally all I have done is make an index.html, main.js. The main.js has your boilerplate createWindow(). It has var usb = require('usb'). I have npm installed electron, electron-rebuild, and node-usb. They are all installed. I reinstalled all of them. I used electron rebuild without an issue, then I tried to run the program and an alert popped with that output. Is node-usb native 32bit?

rwaldron, 2016-09-20:
Ok, that might be enough to get me started. I'll ping back if I have any further questions :)

rwaldron, 2016-09-20:
Using Windows 10 (64-bit), here's what I've done: 

1. `git clone https://github.com/electron/electron-quick-start`
2. `cd electron-quick-start`
3. `npm install`
4. `npm install usb` //  completes successfully
5. `npm install --save-dev electron-rebuild`
  - This won't install for me. Looks like it's failing on nslog :\

woodev01, 2016-09-20:
Yeah, that install takes some work. You need the C++ file from Visual Studio 13 and you need Python 2.7

rwaldron, 2016-09-20:
Ok, I'll take another run at it.

BlackWolf, 2016-10-01:
I'm trying to do the same thing and I am running into 

`Uncaught Error: Cannot find module 'C:\path\to\my\project\node_modules\usb\src\binding\usb_bindings.node`

which I assume is a similar issue. I do have python2.7, but what does "the C++ file from Visual Studio 13" refer to?

thanks!

digitalentity, 2016-11-23:
I have the same issue. Win7 x64, app crashes at `var usb = require('usb');` with the same message. @rwaldron any suggestion where to dig?

Croxy, 2017-02-09:
Has anyone been able to resolve this?

I am getting the same error related to the USB package as above. 

[image]

eohan, 2017-04-04:
Hi, Has anyone been able to resolve this ?

rwaldron, 2017-04-04:
Electron's docs describe additional steps needed when using compiled modules: https://github.com/electron/electron/blob/master/docs/tutorial/using-native-node-modules.md

acr14, 2017-07-18:
I'm running into the same issue and was wondering if anyone has found a solution?

eohan, 2017-07-21:
@acr14 , https://github.com/tessel/node-usb/issues/140#issuecomment-291583363 worked for me.

chenamy2017, 2018-09-18:
I used electron-rebuild resolve this .

closed 2018-10-27 by kevinmehall

vikjsdev, 2020-07-06:
Does anybody found solution?