node-usb/node-usb #141: Cannot build for electron on Win10

BlackWolf, 2016-10-01:
Hello,

I am currently trying to build this module for electron on Windows 10 (VS 2015), but I am unsuccessful so far :/

I tried following the steps from https://github.com/sandeepmistry/noble/issues/235, but I get build errors when I try to rebuild node-usb for electron. This is regardless of if I build for 64 or 32 bit (32bit is what I eventually want).

Any help is appreciated.

```
C:\projects\node_modules\usb>node-gyp rebuild --target=1.3.6 --arch=x64 --dist=http://atom.io/download/atom-shell
gyp info it worked if it ends with ok
gyp info using node-gyp@3.4.0
gyp info using node@6.7.0 | win32 | x64
gyp info spawn C:\Python27\python.exe
gyp info spawn args [ 'C:\\Users\\marsc\\AppData\\Roaming\\npm\\node_modules\\node-gyp\\gyp\\gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'msvs',
gyp info spawn args   '-G',
gyp info spawn args   'msvs_version=auto',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\projects\\DesktopClient\\build\\node_modules\\usb\\build\\config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\marsc\\AppData\\Roaming\\npm\\node_modules\\node-gyp\\addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\marsc\\.node-gyp\\iojs-1.3.6\\common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=C:\\Users\\marsc\\.node-gyp\\iojs-1.3.6',
gyp info spawn args   '-Dnode_gyp_dir=C:\\Users\\marsc\\AppData\\Roaming\\npm\\node_modules\\node-gyp',
gyp info spawn args   '-Dnode_lib_file=iojs.lib',
gyp info spawn args   '-Dmodule_root_dir=C:\\projects\\DesktopClient\\build\\node_modules\\usb',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'C:\\projects\\DesktopClient\\build\\node_modules\\usb\\build',
gyp info spawn args   '-Goutput_dir=.' ]
gyp info spawn C:\Program Files (x86)\MSBuild\14.0\bin\msbuild.exe
gyp info spawn args [ 'build/binding.sln',
gyp info spawn args   '/clp:Verbosity=minimal',
gyp info spawn args   '/nologo',
gyp info spawn args   '/p:Configuration=Release;Platform=x64' ]
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  core.c
  descriptor.c
  hotplug.c
  io.c
C:\projects\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C
2011: 'timespec': 'struct' type redefinition (compiling source file ..\libusb\libusb\io.c) [C:\p
rojects\DesktopClient\build\node_modules\usb\build\libusb.vcxproj]
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: see declara
  tion of 'timespec' (compiling source file ..\libusb\libusb\io.c)
C:\projects\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C
2011: 'timespec': 'struct' type redefinition (compiling source file ..\libusb\libusb\descriptor.
c)..\libusb\libusb\io.c(1227): warning C4244: '=': conversion from 'time_t' to 'long', possible
loss of data [C:\projects\node_modules\usb\build\libusb.vcxproj]

  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: see declara
  tion of 'timespec' (compiling source file ..\libusb\libusb\descriptor.c)
..\libusb\libusb\io.c(1902): warning C4244: '=': conversion from 'time_t' to 'long', possible lo
ss of data [C:\projects\node_modules\usb\build\libusb.vcxproj]
..\libusb\libusb\io.c(2435): warning C4244: '=': conversion from 'time_t' to 'long', possible lo
ss of data [C:\projects\node_modules\usb\build\libusb.vcxproj]
C:\projects\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C
2011: 'timespec': 'struct' type redefinition (compiling source file ..\libusb\libusb\hotplug.c)
[C:\projects\node_modules\usb\build\libusb.vcxproj]
  strerror.c
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: see declara
  tion of 'timespec' (compiling source file ..\libusb\libusb\hotplug.c)
  sync.c
C:\projects\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C
2011: 'timespec': 'struct' type redefinition (compiling source file ..\libusb\libusb\core.c) [C:
\projects\DesktopClient\build\node_modules\usb\build\libusb.vcxproj]
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: see declara
  tion of 'timespec' (compiling source file ..\libusb\libusb\core.c)
  poll_windows.c
  threads_windows.c
C:\projects\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C
2011: 'timespec': 'struct' type redefinition (compiling source file ..\libusb\libusb\sync.c)C:\p
rojects\DesktopClient\build\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011
: 'timespec': 'struct' type redefinition (compiling source file ..\libusb\libusb\strerror.c) [C:
\projects\DesktopClient\build\node_modules\usb\build\libusb.vcxproj]

  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: see declara
  tion of 'timespec' (compiling source file ..\libusb\libusb\sync.c)C:\Program Files (x86)\Windo
  ws Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: see declaration of 'timespec' (compilin
  g source file ..\libusb\libusb\strerror.c)

  windows_usb.c
C:\projects\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C
2011: 'timespec': 'struct' type redefinition (compiling source file ..\libusb\libusb\os\threads_
windows.c) [C:\projects\node_modules\usb\build\libusb.vcxproj]
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: see declara
  tion of 'timespec' (compiling source file ..\libusb\libusb\os\threads_windows.c)
..\libusb\libusb\os\threads_windows.c(194): warning C4244: '=': conversion from 'time_t' to 'lon
g', possible loss of data [C:\projects\node_modules\usb\build\libusb.vcxproj
]
..\libusb\libusb\os\threads_windows.c(196): warning C4244: '=': conversion from 'const time_t' t
o 'long', possible loss of data [C:\projects\node_modules\usb\build\libusb.v
cxproj]
C:\projects\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C
2011: 'timespec': 'struct' type redefinition (compiling source file ..\libusb\libusb\os\poll_win
dows.c) [C:\projects\node_modules\usb\build\libusb.vcxproj]
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: see declara
  tion of 'timespec' (compiling source file ..\libusb\libusb\os\poll_windows.c)
C:\projects\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C
2011: 'timespec': 'struct' type redefinition (compiling source file ..\libusb\libusb\os\windows_
usb.c) [C:\projects\node_modules\usb\build\libusb.vcxproj]
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: see declara
  tion of 'timespec' (compiling source file ..\libusb\libusb\os\windows_usb.c)
..\libusb\libusb\os\windows_usb.c(586): warning C4244: '=': conversion from '__int64' to 'unsign
ed int', possible loss of data [C:\projects\node_modules\usb\build\libusb.vc
xproj]
..\libusb\libusb\os\windows_usb.c(1017): warning C4996: 'GetVersionExA': was declared deprecated
 [C:\projects\node_modules\usb\build\libusb.vcxproj]
  C:\Program Files (x86)\Windows Kits\8.1\Include\um\sysinfoapi.h(433): note: see declaration of
   'GetVersionExA'
..\libusb\libusb\os\windows_usb.c(1020): warning C4996: 'GetVersionExA': was declared deprecated
 [C:\projects\node_modules\usb\build\libusb.vcxproj]
  C:\Program Files (x86)\Windows Kits\8.1\Include\um\sysinfoapi.h(433): note: see declaration of
   'GetVersionExA'
..\libusb\libusb\os\windows_usb.c(1423): warning C4244: '=': conversion from '__int64' to 'unsig
ned int', possible loss of data [C:\projects\node_modules\usb\build\libusb.v
cxproj]
..\libusb\libusb\os\windows_usb.c(2558): warning C4018: '<': signed/unsigned mismatch [C:\projec
ts\DesktopClient\build\node_modules\usb\build\libusb.vcxproj]
gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\MSBuild\14.0\bin\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Users\marsc\AppData\Roaming\npm\node_modules\node-gyp\lib\build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Windows_NT 10.0.14393
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\marsc\\AppData\\Roaming\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild" "--target=1.3.6" "--arch=x64" "--dist=http://atom.io/download/atom-shell"
gyp ERR! cwd C:\projects\node_modules\usb
gyp ERR! node -v v6.7.0
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok

```

tcr, 2016-10-05:
Hi, can you try the suggestion in this thread and let me know if it fixes it for you? https://github.com/libusb/libusb/issues/144

digital-flowers, 2016-10-17:
instead of changing libusb source code to fix this you can just edit libusb.gypi and define for both `'HAVE_STRUCT_TIMESPEC=1','_TIMESPEC_DEFINED=1'` for windows target

digital-flowers, 2016-10-17:
for electron i already did this so until this bug fixed you can use my fork, just do:
`npm install git+https://git@github.com/digital-flowers/electron-usb.git`

BlackWolf, 2016-10-18:
Removing the typestruc lines indeed helped :-) Thanks for all the suggestions nonetheless.

wkentdag, 2016-11-11:
@digital-flowers solution **finally** got me past this error after I tried every solution I could find, including everything in [this thread](https://discuss.atom.io/t/use-node-usb-inside-electron/16748/2). Thank you!

wkentdag, 2016-12-03:
update: one of my coworkers has created a fork that has the requisite updates to `libusb.gypi`: https://github.com/aktary/node-usb

We made it because, while this helped us get the module working in dev mode, we ran into issues in production. Most of our modules are getting bundled up by webpack during the build process, but webpack (unsurprisingly) throws an error when it tries to bundle up the `usb` module. Our solution was to add `usb` as a webpack external...but that means that it needs to be `npm install`'d by electron-builder during the build process....which fails because the fresh install of `node-usb` is missing the edits that @digital-flowers originally pointed out. 

So with this fork, we're able to successfully deploy to production by just `npm i git+https://github.com/aktary/node-usb` and then running `electron-rebuild -w usb -p`. Hope this module might help some others avoid this tedious error 😄 . 

Also, it makes me wonder if there's a good reason why the `HAVE_STRUCT_TIMESPEC` and `_TIMESPEC_DEFINED` options aren't set here by default...

sebakerckhof, 2017-01-06:
@wkentdag thanks, this came in real handy today.

Good4m, 2017-01-15:
@wkentdag I really appreciate it man. Thank you so much.

jrobeson, 2019-03-19:
Fixed in https://github.com/hetsch/node-usb/pull/1

kevinmehall, 2019-05-19:
Closing all old Electron build related issues now that we have prebuilds for Electron in 1.6.0.

closed 2019-05-19 by kevinmehall