node-usb/node-usb #147: LIBUSB_ERROR_NOT_SUPPORTED problem on Windows

fjozsef, 2016-10-29:
Hi!

I have the following algorithm:
```node
#!/usr/bin/env node
'use strict';

var usb = require('usb');

// replaced with the correct ids
var vid = 0x0000;
var pid = 0x0000;

var device = usb.findByIds(vid, pid);
device.open();

var usbInterface = device.interface(0);
if (usbInterface.isKernelDriverActive()) {
    usbInterface.detachKernelDriver();
}
```

The problem is that at `usbInterface.isKernelDriverActive()` line throws on windows 10, but it works perfectly on Linux ubuntu.
Error:

```bash
C:\work\usb\node_modules\usb\usb.js:206
        return this.device.__isKernelDriverActive(this.id)
                           ^

Error: LIBUSB_ERROR_NOT_SUPPORTED
    at Error (native)
    at Interface.isKernelDriverActive (C:\work\usb\node_modules\usb\usb.js:206:21)
```

I've downloaded Zadig 2.2 from [here](http://zadig.akeo.ie/) and installed WinUSB, but the problem persisted.

Could you help what I am doing wrong?

kevinmehall, 2016-10-29:
http://libusb.sourceforge.net/api-1.0/group__dev.html#gab14d11ed6eac7519bb94795659d2c971
> This functionality is not available on Windows.

You probably don't need the isKernelDriverActive / detachKernelDriver even on Linux, as most devices you'd use libusb for don't have kernel drivers.

closed 2016-10-29 by kevinmehall

sairam1919, 2017-04-24:
Hi ,

I have the following code

var usb = require('usb');
term = usb.findByIds(1921,21904);
console.log(term);
term.open();

when i try to run the code i am getting the following error

  this.__open()
             ^

Error: LIBUSB_ERROR_NOT_SUPPORTED
    at Error (native)
    at Device.usb.Device.open (C:\Users\PO323106\Desktop\unity_poc\node_modules\usb\usb.js:33:7)
    at Object.<anonymous> (C:\Users\PO323106\Desktop\unity_poc\usb.js:10:6)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.runMain (module.js:604:10)
    at run (bootstrap_node.js:394:7)

what is the issue with this.

ivankennethwang, 2018-09-20:
@sairam1919 did you manage to fix this issue?

Chris2011, 2020-11-04:
Check whether you have the correct driver for your device. I listed all my devices in Zadig and my flash storage had a different one. After I've overwritten it with WinUSB it worked.