node-usb/node-usb #150: Can't build from source. Help!

k1tzu, 2016-11-11:
Pre-built version now stucks silently from time to time so I'm trying to install it from source to get the root cause.
$ npm install

> usb@1.2.2 install /home/node-usb
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download(404): https://github.com/tessel/node-usb/releases/download/1.2.2/usb_bindings-v1.2.2-node-v51-linux-arm.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for usb@1.2.2 and node@7.1.0 (node-v51 ABI) (falling back to source compile with node-gyp) 
make: Entering directory '/home/node-usb/build'
make: *** No rule to make target 'Release/obj.target/libusb/libusb/libusb/core.o', needed by 'Release/obj.target/usb.a'.  Stop.
make: Leaving directory '/home/node
...

NirLevanon, 2016-12-01:
Similar:

```
npm WARN onload-script Error: Cannot find module 'undefined'
npm WARN onload-script     at Function.Module._resolveFilename (module.js:325:15)
npm WARN onload-script     at Function.Module._load (module.js:276:25)
npm WARN onload-script     at Module.require (module.js:353:17)
npm WARN onload-script     at require (internal/module.js:12:17)
npm WARN onload-script     at cb (C:\Development\emulator\node_modules\npm\lib\npm.js:199:11)
npm WARN onload-script     at C:\Development\emulator\node_modules\npm\lib\npm.js:304:16
npm WARN onload-script     at C:\Development\emulator\node_modules\npm\lib\config\core.js:81:7
npm WARN onload-script     at Array.forEach (native)
npm WARN onload-script     at C:\Development\emulator\node_modules\npm\lib\config\core.js:80:13
npm WARN onload-script     at f (C:\Development\emulator\node_modules\npm\node_modules\once\once.js:25:25)
npm WARN onload-script  { [Error: Cannot find module 'undefined'] code: 'MODULE_NOT_FOUND' }
node-pre-gyp ERR! Tried to download(404): https://github.com/node-hid/node-hid/releases/download/0.5.2/HID-v0.5.2-electron-v1.4-win32-ia32.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for node-hid@0.5.2 and electron@1.4.10 (electron-v1.4 ABI) (falling back to source compile with node-gyp)
npm WARN prefer global npm@3.10.10 should be installed with -g
node-pre-gyp ERR! Tried to download: https://github.com/tessel/node-usb/releases/download/1.2.0/usb_bindings-v1.2.0-electron-v1.4-win32-ia32.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for usb@1.2.0 and electron@1.4.10 (electron-v1.4 ABI) (falling back to source compile with node-gyp)
1.2.0 and electron@1.4.10 (electron-v1.4 ABI) (falling back to source compile with node-gyp)
```

rimmartin, 2016-12-03:
From github you can use the --recursive switch to get the libusb submodule to download into the libusb folder

```
git clone --recursive https://github.com/tessel/node-usb.git node-usb
```
then cd into node-usb and 

```
npm install
npm test
```

closed 2016-12-09 by k1tzu

xp6530, 2017-06-05:
so why?