node-usb/node-usb #152: LIBUSB_ERROR_ACCESS  this.__open()

pawanpec, 2016-11-17:
Hi,
I am getting below error while reading USB.
     this.__open()
             ^

Error: LIBUSB_ERROR_ACCESS
    at Error (native)
    at Device.usb.Device.open (/node_modules/usb/usb.js:33:7)
    at /scripts/usbtest.js:10:10
    at Array.forEach (native)
    at Object.<anonymous> i/scripts/usbtest.js:7:9)
    at Module._compile (module.js:556:32)
    at Object.Module._extensions..js (module.js:565:10)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)

var usb = require('usb');
var devices = usb.getDeviceList();
//var device = devices[0];
devices.forEach(function(device){
  console.log(device);
  console.log(device.deviceDescriptor.idProduct);
  device.open();
  console.log("device.interfaces"+ device.interfaces);

BrutalFaust, 2016-11-23:
Hello,

I have the same problem, can you tell me if you was could solve that mistake?

thank you in advance.

rimmartin, 2016-12-03:
you need user permissions for libusb

kevinmehall, 2017-05-20:
See also #174, or else your user account doesn't have access to open USB devices.

closed 2017-05-20 by kevinmehall