node-usb/node-usb #155: 'Cannot read property 'call' of undefined at usb.js

Paul-Ver, 2016-11-26:
I've installed nuimo.js on my windows 10 PC, which uses the Noble BLE library by Sandeep Mistry.
It works fine on Mac, but not on Windows 10.

My little test program will work fine, I've set up my bluetooth device with libusb and it all runs well (connects), but after some seconds, it will crash with the following error: 

```
D:\Hobby-Work\node\Bridge\node_modules\usb\usb.js:113
                process.nextTick(function() { callback.call(self, e); });
                                                      ^

TypeError: Cannot read property 'call' of undefined
    at D:\Hobby-Work\node\Bridge\node_modules\usb\usb.js:113:41
    at _combinedTickCallback (internal/process/next_tick.js:67:7)
    at process._tickCallback (internal/process/next_tick.js:98:9)
```

Paul-Ver, 2016-11-26:
I've added a check:

```
if(callback){
   process.nextTick(function() { callback.call(self, e); });
}
```

Which seems to work, but there may be a problem with your code.
It seems like you're expecting a callback, but not giving one, somewhere in your code.
(Or it's optional, but in that case you don't check if it's defined)

Also, the piece of code is in a catch statement, I've read that you shouldn't base your code logic on exceptions, but I'm not sure if that's the case here.

kevinmehall, 2017-02-18:
You should pass a callback to `controlTransfer`.

closed 2017-02-18 by kevinmehall

codenakama, 2019-08-27:
Hi,

I am getting this error too when doing

```
device.open();
  device.setConfiguration(1);
  device.interface(0).claim();
  device
    .interface(0)
    .endpoint(1)
    .transfer(frameHeader, err => {
      if (!err) {
        device.interface(0).claim();
        device
          .interface(0)
          .endpoint(1)
          .transfer(data, err => console.dir(err));
      }
    });
```

is there anything particularly wrong that I am doing?

thanks in advance.

PR #201 (2017-12-02): Make callback argument in controlTransfer completely optional
[Line 112](https://github.com/tessel/node-usb/blob/master/usb.js#L112) suggests that the callback argument to the [controlTransfer](https://github.com/tessel/node-usb/blob/master/usb.js#L79) function is optional.

However, in case of an error the callback is [called](https://github.com/tessel/node-usb/blob/master/usb.js#L125) unconditionally.

Fixes #155