node-usb/node-usb #157: Electron OSX: Module version mismatch. Expected 50, got 51.

catskull, 2016-12-10:
I'm using node v 7.1.0 on macOS Sierra, electron-prebuilt 1.4.1 and node-usb 1.2.0. After doing npm install, npm start gives me the error "Module version mismatch. Expected 50, got 51."

I have tried going into node_modules/usb and running `npm rebuild --runtime=electron --target=1.4.1 --disturl=https://atom.io/download/atom-shell --abi=50` to no avail.

Sorry I'm new to node/electron, is there an easy fix for this?

ert78gb, 2017-06-28:
check electron-rebuild package

catskull, 2017-08-05:
After fiddling around quite a bit, that seems to have solved the issue!

For those playing along at home, here's the package: https://github.com/electron/electron-rebuild

I thought I'd be cool and use `yarn`, but don't do that. I had to remove `node_modules`, `npm install` and then run `node_modules/.bin/electron-rebuild` and it worked!

closed 2017-08-05 by catskull