node-usb/node-usb #158: Ensure builds for Node 6.x.x

rwaldron, 2016-12-19:
I discovered a poor review on Sparkfun.com which blamed a failed linux installation. The reviews are here: https://www.sparkfun.com/products/13847 Unfortunately, there is no way to link directly to the review, so click on "Reviews" and scroll to the review titled "Software Not Ready for Prime Time"

OwenBrotherwood, 2017-02-18:
node-usb is also in problems with Windows 10 

As node-usb has problems with https://github.com/tessel/node-usb/issues/163 node 7, I have had to go back to node -v v6.8.1 in order to get a version of nw.js to work (using last released version of nw.js that is v6.x.x).

I now have to go and ensure that the requirements are met in connection with build tools, however, all other binding.gyp in the Open Source project work EXCEPT for node-usb.

```
C:\source\repos\github\espruino\EspruinoWebIDE\node_modules>ls -R */binding.gyp
bluetooth-hci-socket/binding.gyp  serialport/binding.gyp  usb/binding.gyp  websocket/binding.gyp  winnus/binding.gyp
```
## 
Is there anything I can do to help with the issue(s)?

```
gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\MSBuild\14.0\bin\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Windows_NT 10.0.14393
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "build" "--fallback-to-build" "--module=C:\\source\\repos\\github\\tessel\\node-usb\\src\\binding\\usb_bindings.node" "--module_name=usb_bindings" "--module_path=C:\\source\\repos\\github\\tessel\\node-usb\\src\\binding"
gyp ERR! cwd C:\source\repos\github\tessel\node-usb
gyp ERR! node -v v6.8.1
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute 'node-gyp.cmd build --fallback-to-build --module=C:\source\repos\github\tessel\node-usb\src\binding\usb_bindings.node --module_name=usb_bindings --module_path=C:\source\repos\github\tessel\node-usb\src\binding' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (C:\source\repos\github\tessel\node-usb\node_modules\node-pre-gyp\lib\util\compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:106:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:877:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
node-pre-gyp ERR! System Windows_NT 10.0.14393
node-pre-gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\source\\repos\\github\\tessel\\node-usb\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd C:\source\repos\github\tessel\node-usb
node-pre-gyp ERR! node -v v6.8.1
node-pre-gyp ERR! node-pre-gyp -v v0.6.33
node-pre-gyp ERR! not ok
Failed to execute 'node-gyp.cmd build --fallback-to-build --module=C:\source\repos\github\tessel\node-usb\src\binding\usb_bindings.node --module_name=usb_bindings --module_path=C:\source\repos\github\tessel\node-usb\src\binding' (1)

npm ERR! Windows_NT 10.0.14393
npm ERR! argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "install"
npm ERR! node v6.8.1
npm ERR! npm  v3.10.8
npm ERR! code ELIFECYCLE
npm ERR! usb@1.2.2 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1

```

kevinmehall, 2017-02-18:
@OwenBrotherwood, this package is `usb` on npm, not `node-usb`

OwenBrotherwood, 2017-02-18:
Aaarg, the npm module is a beautiful piece of work: re-evaluation of testing and loss of face giving a hour or two of double check.
(It was a quick test to see if my build environment worked:  it did, if I just spelt right)

OwenBrotherwood, 2017-02-18:
with a correctly spelt:

```
npm init
npm install usb # which went well
```

I now have to go find something that can do something with these results.
NB:
Addition of following in variables section of binding.gyp, otherwise another error.
The correctness of these are unknown, as they where obtained from a comment in another issue

```
    'module_name': 'usb_bindings', 
    'module_path': './src/usb_binding'
```

```
> C:\source\test\node_modules\usb> node-gyp rebuild
> gyp info it worked if it ends with ok
> gyp info using node-gyp@3.5.0
> gyp info using node@6.8.1 | win32 | x64
> gyp info spawn C:\Python27\python.EXE
> gyp info spawn args [ 'C:\\Users\\owen\\AppData\\Roaming\\npm\\node_modules\\node-gyp\\gyp\\gyp_main.py',
> gyp info spawn args   'binding.gyp',
> gyp info spawn args   '-f',
> gyp info spawn args   'msvs',
> gyp info spawn args   '-G',
> gyp info spawn args   'msvs_version=auto',
> gyp info spawn args   '-I',
> gyp info spawn args   'C:\\source\\test\\node_modules\\usb\\build\\config.gypi',
> gyp info spawn args   '-I',
> gyp info spawn args   'C:\\Users\\owen\\AppData\\Roaming\\npm\\node_modules\\node-gyp\\addon.gypi',
> gyp info spawn args   '-I',
> gyp info spawn args   'C:\\Users\\owen\\.node-gyp\\6.8.1\\include\\node\\common.gypi',
> gyp info spawn args   '-Dlibrary=shared_library',
> gyp info spawn args   '-Dvisibility=default',
> gyp info spawn args   '-Dnode_root_dir=C:\\Users\\owen\\.node-gyp\\6.8.1',
> gyp info spawn args   '-Dnode_gyp_dir=C:\\Users\\owen\\AppData\\Roaming\\npm\\node_modules\\node-gyp',
> gyp info spawn args   '-Dnode_lib_file=node.lib',
> gyp info spawn args   '-Dmodule_root_dir=C:\\source\\test\\node_modules\\usb',
> gyp info spawn args   '--depth=.',
> gyp info spawn args   '--no-parallel',
> gyp info spawn args   '--generator-output',
> gyp info spawn args   'C:\\source\\test\\node_modules\\usb\\build',
> gyp info spawn args   '-Goutput_dir=.' ]
> gyp info spawn C:\Program Files (x86)\MSBuild\14.0\bin\msbuild.exe
> gyp info spawn args [ 'build/binding.sln',
> gyp info spawn args   '/clp:Verbosity=minimal',
> gyp info spawn args   '/nologo',
> gyp info spawn args   '/p:Configuration=Release;Platform=x64' ]
> Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
>   core.c
>   descriptor.c
>   hotplug.c
>   io.c
> C:\source\test\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: 'timespec': 'struct' type redefini
> tion (compiling source file ..\libusb\libusb\descriptor.c) [C:\source\test\node_modules\usb\build\libusb.vcxproj]
>   C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: see declaration of 'timespec' (com
>   piling source file ..\libusb\libusb\descriptor.c)C:\source\test\node_modules\usb\libusb\libusb\os/threads_windows.h(4
>   7): error C2011: 'timespec': 'struct' type redefinition (compiling source file ..\libusb\libusb\io.c)
> 
>   C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: see declaration of 'timespec' (com
>   piling source file ..\libusb\libusb\io.c)
> ..\libusb\libusb\io.c(1227): warning C4244: '=': conversion from 'time_t' to 'long', possible loss of data [C:\source\t
> est\node_modules\usb\build\libusb.vcxproj]
> ..\libusb\libusb\io.c(1902): warning C4244: '=': conversion from 'time_t' to 'long', possible loss of data [C:\source\t
> est\node_modules\usb\build\libusb.vcxproj]
> ..\libusb\libusb\io.c(2435): warning C4244: '=': conversion from 'time_t' to 'long', possible loss of data [C:\source\t
> est\node_modules\usb\build\libusb.vcxproj]
[... 161 lines cut ...]
of data [C:\source\test\node_modules\usb\build\libusb.vcxproj]
..\libusb\libusb\os\windows_usb.c(1017): warning C4996: 'GetVersionExA': was declared deprecated [C:\source\test\node_m
odules\usb\build\libusb.vcxproj]
  C:\Program Files (x86)\Windows Kits\8.1\Include\um\sysinfoapi.h(433): note: see declaration of 'GetVersionExA'
..\libusb\libusb\os\windows_usb.c(1020): warning C4996: 'GetVersionExA': was declared deprecated [C:\source\test\node_m
odules\usb\build\libusb.vcxproj]
  C:\Program Files (x86)\Windows Kits\8.1\Include\um\sysinfoapi.h(433): note: see declaration of 'GetVersionExA'
..\libusb\libusb\os\windows_usb.c(1423): warning C4244: '=': conversion from '__int64' to 'unsigned int', possible loss
 of data [C:\source\test\node_modules\usb\build\libusb.vcxproj]
..\libusb\libusb\os\windows_usb.c(2558): warning C4018: '<': signed/unsigned mismatch [C:\source\test\node_modules\usb\
build\libusb.vcxproj]
gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\MSBuild\14.0\bin\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Users\owen\AppData\Roaming\npm\node_modules\nw-gyp\lib\build.js:294:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Windows_NT 10.0.14393
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\owen\\AppData\\Roaming\\npm\\node_modules\\nw-gyp\\bin\\nw-gyp.js" "rebuild" "--target=0.18.6" "--arch=x64" "--msvs_version=2015"
gyp ERR! cwd C:\source\test\node_modules\usb
gyp ERR! node -v v6.8.1
gyp ERR! nw-gyp -v v3.4.0
gyp ERR! not ok

```

kevinmehall, 2017-02-18:
This is #162 / #109, and you can test https://github.com/tessel/node-usb/pull/165 to see if it solves the VS2015 compilation error.

kevinmehall, 2017-02-18:
@rwaldron, it works for me on 6.x on Linux. The user on Sparkfun only included the NPM part of the error message, not the actual compile error, so this is not actionable. It's probably something like not having a C++ toolchain installed, which should probably be added to the T2 Linux install instructions.

closed 2017-02-18 by kevinmehall