node-usb/node-usb #159: Soft Disconnect-Reconnect via Node USB

hhff, 2016-12-27:
Hi All!

Thanks a ton for the hard work!  It's a great lib.

I'm working on software for a USB Device that requires a "soft disconnect" then "soft reconnect" so that the Device software can perform internal updates.

Is there a way to do this with Node USB?  I've tried `reset` and `Interface#release` to no avail.

Thanks so much!

kevinmehall, 2016-12-27:
`Device#reset` should be enough for a device that restarts itself after a firmware update. Note that OSX >= 10.11 [silently ignores](https://sourceforge.net/p/libusb/mailman/message/34744234/) reset commands from libusb, though.

If you need to cut power to the device instead of resetting its USB controller, your options are more limited. You can send the hub upstream of the device commands to toggle power if it supports that (most hubs, especially root hubs don't) with https://github.com/tcr/node-ppps.

closed 2016-12-27 by kevinmehall

gniezen, 2018-07-26:
@kevinmehall I've just reported `ResetDevice` not working upstream, as I couldn't find a mention of it in the GitHub issues (just the thread from 2016 you linked to). It seems that node-usb uses a fork of libusb. Wouldn't it be quicker/easier to implement the one-line fix there?

kevinmehall, 2018-07-26:
I don't think `usbDeviceReEnumerate` is an exact replacement, as it doesn't perform the reset signalling. So it would be confusing to make `reset` not perform a reset, even if in many applications it's only used for the side effect of re-enumerating. So it may want to be a new libusb function / node-usb method, though AFAIK there is not an equivalent on other platforms.

semireg, 2019-04-18:
You might be interested to see https://github.com/tessel/node-usb/issues/310 that describes how device.setConfiguration works as a device reset.

From the libusb docs:
```If you call this function on a device already configured with the selected configuration, then this function will act as a lightweight device reset: it will issue a SET_CONFIGURATION request using the current configuration, causing most USB-related device state to be reset (altsetting reset to zero, endpoint halts cleared, toggles reset).```

It worked for me!

gniezen, 2019-06-21:
Unfortunately it doesn't look like the "`setConfiguration` as device reset" trick works for my use case, where the TUSB3410 chip requires a device reset after having its firmware written into it.