node-usb/node-usb #162: Trying to rebuild node-usb for Electron on Windows 10

fdeitelhoff, 2017-01-24:
I'm trying to rebuild node-usb for my Electron application on Windows 10. I install node-usb, node-gyp and tried the following command:

`E:\Masterthesis\Simplex-Electron\node_modules\usb>node-gyp rebuild --target=1.4.14 --dist-url=https://atom.io/download/atom-shell`

Unfortunately I get the following output with an error at the end:

```
gyp info it worked if it ends with ok
gyp info using node-gyp@3.4.0
gyp info using node@4.4.5 | win32 | x64
gyp http GET https://atom.io/download/atom-shell/v1.4.14/iojs-v1.4.14.tar.gz
gyp http 200 https://atom.io/download/atom-shell/v1.4.14/iojs-v1.4.14.tar.gz
gyp http GET https://atom.io/download/atom-shell/v1.4.14/SHASUMS256.txt
gyp http GET https://atom.io/download/atom-shell/v1.4.14/win-x64/iojs.lib
gyp http GET https://atom.io/download/atom-shell/v1.4.14/win-x86/iojs.lib
gyp http 200 https://atom.io/download/atom-shell/v1.4.14/win-x86/iojs.lib
gyp http 200 https://atom.io/download/atom-shell/v1.4.14/SHASUMS256.txt
gyp http 200 https://atom.io/download/atom-shell/v1.4.14/win-x64/iojs.lib
gyp info spawn C:\Python27\python.exe
gyp info spawn args [ 'C:\\Users\\Fabia\\AppData\\Roaming\\npm\\node_modules\\node-gyp\\gyp\\gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'msvs',
gyp info spawn args   '-G',
gyp info spawn args   'msvs_version=auto',
gyp info spawn args   '-I',
gyp info spawn args   'E:\\Masterthesis\\Simplex-Electron\\node_modules\\usb\\build\\config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\Fabia\\AppData\\Roaming\\npm\\node_modules\\node-gyp\\addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\Fabia\\.node-gyp\\iojs-1.4.14\\common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=C:\\Users\\Fabia\\.node-gyp\\iojs-1.4.14',
gyp info spawn args   '-Dnode_gyp_dir=C:\\Users\\Fabia\\AppData\\Roaming\\npm\\node_modules\\node-gyp',
gyp info spawn args   '-Dnode_lib_file=iojs.lib',
gyp info spawn args   '-Dmodule_root_dir=E:\\Masterthesis\\Simplex-Electron\\node_modules\\usb',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'E:\\Masterthesis\\Simplex-Electron\\node_modules\\usb\\build',
gyp info spawn args   '-Goutput_dir=.' ]
gyp info spawn C:\Program Files (x86)\MSBuild\14.0\bin\msbuild.exe
gyp info spawn args [ 'build/binding.sln',
gyp info spawn args   '/clp:Verbosity=minimal',
gyp info spawn args   '/nologo',
gyp info spawn args   '/p:Configuration=Release;Platform=x64' ]
Die Projekte in dieser Projektmappe werden nacheinander erstellt. Um eine parallele Erstellung zu ermöglichen, müssen Sie den Schalter "/m" hinzufügen.
  core.c
  descriptor.c
  hotplug.c
  io.c
  strerror.c
  sync.c
  poll_windows.c
  threads_windows.c
E:\Masterthesis\Simplex-Electron\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: "timespec": "struct" Typneudefinition (Quelldatei wird kompiliert ..\libusb\libusb\sync.c) [E:\Masterthesis\Simplex-Electron\node_modules\usb\build\libusb.vcxpr
oj]
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: Siehe Deklaration von "timespec" (Quelldatei wird kompiliert ..\libusb\libusb\sync.c)
  windows_usb.c
E:\Masterthesis\Simplex-Electron\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: "timespec": "struct" Typneudefinition (Quelldatei wird kompiliert ..\libusb\libusb\io.c) [E:\Masterthesis\Simplex-Electron\node_modules\usb\build\libusb.vcxproj
]
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: Siehe Deklaration von "timespec" (Quelldatei wird kompiliert ..\libusb\libusb\io.c)
E:\Masterthesis\Simplex-Electron\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: "timespec": "struct" Typneudefinition (Quelldatei wird kompiliert ..\libusb\libusb\hotplug.c) [E:\Masterthesis\Simplex-Electron\node_modules\usb\build\libusb.vc
xproj]
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: Siehe Deklaration von "timespec" (Quelldatei wird kompiliert ..\libusb\libusb\hotplug.c)..\libusb\libusb\io.c(1227): warning C4244: "=": Konvertierung von "time_t" in "long", mög
  licher Datenverlust

..\libusb\libusb\io.c(1902): warning C4244: "=": Konvertierung von "time_t" in "long", möglicher Datenverlust [E:\Masterthesis\Simplex-Electron\node_modules\usb\build\libusb.vcxproj]
E:\Masterthesis\Simplex-Electron\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: "timespec": "struct" Typneudefinition (Quelldatei wird kompiliert ..\libusb\libusb\os\threads_windows.c)..\libusb\libusb\io.c(2435): warning C4244: "=": Konvert
[... 18 lines cut ...]
E:\Masterthesis\Simplex-Electron\node_modules\usb\libusb\libusb\os/threads_windows.h(47): error C2011: "timespec": "struct" Typneudefinition (Quelldatei wird kompiliert ..\libusb\libusb\os\windows_usb.c) [E:\Masterthesis\Simplex-Electron\node_modules\usb\build\li
busb.vcxproj]
  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\time.h(39): note: Siehe Deklaration von "timespec" (Quelldatei wird kompiliert ..\libusb\libusb\os\windows_usb.c)
..\libusb\libusb\os\windows_usb.c(586): warning C4244: "=": Konvertierung von "__int64" in "unsigned int", möglicher Datenverlust [E:\Masterthesis\Simplex-Electron\node_modules\usb\build\libusb.vcxproj]
..\libusb\libusb\os\windows_usb.c(1017): warning C4996: 'GetVersionExA': wurde als veraltet deklariert [E:\Masterthesis\Simplex-Electron\node_modules\usb\build\libusb.vcxproj]
  C:\Program Files (x86)\Windows Kits\8.1\Include\um\sysinfoapi.h(433): note: Siehe Deklaration von "GetVersionExA"
..\libusb\libusb\os\windows_usb.c(1020): warning C4996: 'GetVersionExA': wurde als veraltet deklariert [E:\Masterthesis\Simplex-Electron\node_modules\usb\build\libusb.vcxproj]
  C:\Program Files (x86)\Windows Kits\8.1\Include\um\sysinfoapi.h(433): note: Siehe Deklaration von "GetVersionExA"
..\libusb\libusb\os\windows_usb.c(1423): warning C4244: "=": Konvertierung von "__int64" in "unsigned int", möglicher Datenverlust [E:\Masterthesis\Simplex-Electron\node_modules\usb\build\libusb.vcxproj]
..\libusb\libusb\os\windows_usb.c(2558): warning C4018: "<": Konflikt zwischen "signed" und "unsigned" [E:\Masterthesis\Simplex-Electron\node_modules\usb\build\libusb.vcxproj]
gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\MSBuild\14.0\bin\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Users\Fabia\AppData\Roaming\npm\node_modules\node-gyp\lib\build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Windows_NT 10.0.14393
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\Fabia\\AppData\\Roaming\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild" "--target=1.4.14" "--dist-url=https://atom.io/download/atom-shell"
gyp ERR! cwd E:\Masterthesis\Simplex-Electron\node_modules\usb
gyp ERR! node -v v4.4.5
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok
```

Maybe someone had this problem already because I've no clue what I could fix.

Cedware, 2017-02-01:
Seems you have been running into the same problem as me: #109
Actually it has been fixed since around a year, but it should work by using vs 2013, but I haven't tried yet.

Cedware, 2017-02-01:
I created a fork, which solves the problem:
https://github.com/Cedware/node-usb
but I had to update to a newer version of libusb, so I can't promise everything works as expected

fdeitelhoff, 2017-02-06:
@Cedware First of all thank you!

But I'm not quite sure how to proceed in this case. Should I just use your fork? Won't this be a problem with future releases of node-usb? Is it planned to fix the problem in the main repository?

Haven't tried VS 2013 yet and will do this if there's a chance this will be fixed.

Cedware, 2017-02-06:
It should just be a transitional solution, hopefully the project owner is going to fix this issue in the main repository, or he wants me to create a merge request.
As long nothing of this happens I'm going to keep my fork online.
I think it's worth trying it with VS 2013 if you are fine with investing some time for a unknown result.

demshy, 2017-02-12:
I had the same problem and was about to give up, so I couldnt believe when your fork worked @Cedware. Thanks a lot! Hope the project owner acknowledges it.

kevinmehall, 2017-02-15:
Can someone affected by this issue please test https://github.com/tessel/node-usb/pull/165? (You'll have to `git submodule update` after pulling the `vs2015` branch)

OwenBrotherwood, 2017-02-18:
In progress with:

```
git fetch
git checkout vs2015 
```

This comment get's edited until https://en.wikipedia.org/wiki/It_ain't_over_till_the_fat_lady_sings

kevinmehall, 2017-02-18:
@OwenBrotherwood: You will also need to `git submodule update` because the change is in the `libusb` submodule.

OwenBrotherwood, 2017-02-18:
OK: I am back to the enclosed error with node-gyp rebuild and to if the following is correct, the addition of module_name and module_path in variables

```

  'variables': {
    'use_udev%': 1,
    'use_system_libusb%': 'false',
    'module_name': 'usb_bindings', 
    'module_path': './src/usb_binding'
  },

```
```
C:\source\repos\github\tessel\node-usb>node-gyp rebuild
gyp info it worked if it ends with ok
gyp info using node-gyp@3.5.0
gyp info using node@6.8.1 | win32 | x64
gyp info spawn C:\Python27\python.EXE
gyp info spawn args [ 'C:\\Users\\owen\\AppData\\Roaming\\npm\\node_modules\\node-gyp\\gyp\\gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'msvs',
gyp info spawn args   '-G',
gyp info spawn args   'msvs_version=auto',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\source\\repos\\github\\tessel\\node-usb\\build\\config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\owen\\AppData\\Roaming\\npm\\node_modules\\node-gyp\\addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\owen\\.node-gyp\\6.8.1\\include\\node\\common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=C:\\Users\\owen\\.node-gyp\\6.8.1',
gyp info spawn args   '-Dnode_gyp_dir=C:\\Users\\owen\\AppData\\Roaming\\npm\\node_modules\\node-gyp',
gyp info spawn args   '-Dnode_lib_file=node.lib',
gyp info spawn args   '-Dmodule_root_dir=C:\\source\\repos\\github\\tessel\\node-usb',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'C:\\source\\repos\\github\\tessel\\node-usb\\build',
gyp info spawn args   '-Goutput_dir=.' ]
gyp: Undefined variable module_name in binding.gyp while trying to load binding.gyp
gyp ERR! configure error
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (C:\Users\owen\AppData\Roaming\npm\node_modules\node-gyp\lib\configure.js:308:16)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Windows_NT 10.0.14393
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\owen\\AppData\\Roaming\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\source\repos\github\tessel\node-usb
gyp ERR! node -v v6.8.1
gyp ERR! node-gyp -v v3.5.0
gyp ERR! not ok

```

kevinmehall, 2017-02-18:
Try using `./node_modules/.bin/node-pre-gyp rebuild` instead of `node-gyp rebuild`

OwenBrotherwood, 2017-02-18:
much better 👍

OwenBrotherwood, 2017-02-18:
OK: I still am not at my goal but I was able to the basics.
My final tests mean time and getting usb to work with a back level nw.js.

when usb is ready for node 7 then I can use a current nw.js

kevinmehall, 2017-02-18:
@OwenBrotherwood: The `master` branch works with Node 7.x, and probably the most recent release too. #163 is about binary releases that won't help you with node-webkit anyway.

burakkilic, 2017-06-01:
Any update on this? I cannot use the library in Windows with electron.

saboya, 2017-06-24:
Version 1.3.0 was tagged recently, you can change your package.json to use that:

```
...
    "usb": "git+https://github.com/tessel/node-usb.git#1.3.0",
...
```

bglase, 2017-10-17:
What is the timeframe for publishing this to NPM?

gniezen, 2017-11-13:
@kevinmehall Is there anything we can do to help get this released to npm? We're using yarn which doesn't support submodules in git dependencies.

gniezen, 2017-12-19:
v1.3.1 that's now on npm builds on VS2017, thank you!

jrobeson, 2019-03-01:
This ticket can probably be closed now right?

harshadakulkarni, 2019-04-24:
use command npm install electron-usb@0.0.11

kevinmehall, 2019-05-19:
Closing all old Electron build related issues now that we have prebuilds for Electron in 1.6.0.

closed 2019-05-19 by kevinmehall