node-usb/node-usb #163: Failing to install on Windows

endquote, 2017-02-13:
I'm having this issue...

```
node-pre-gyp ERR! Tried to download: https://github.com/tessel/node-usb/releases/download/1.2.0/usb_bindings-v1.2.0-node-v51-win32-x64.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for usb@1.2.0 and node@7.2.1 (node-v51 ABI) (falling back to source compile with node-gyp)
```

So the `v51` file doesn't exist, but `v48` does. What is that version number based on? I tried a few different versions of node but always got the same result.

I also have VS2015 and [windows-build-tools](https://github.com/felixrieseberg/windows-build-tools) installed, but the native compile didn't work either.

kevinmehall, 2017-02-13:
The ABI version is the NODE_MODULE_VERSION https://nodejs.org/en/download/releases/. There are no pre-built node-usb binaries for v7.x yet; I'm updating this issue to track that.

endquote, 2017-02-13:
Understood, it works if I drop down to 6.9.5. (I thought I'd tried that, but turns out my nvm-windows setup was broken.)

kevinmehall, 2017-12-04:
1.4.0 released with binaries for Node 7, 8, and 9

closed 2017-12-04 by kevinmehall