node-usb/node-usb #166: Missing functions on usb object - noob question

daysleeper29, 2017-02-18:
Hi, when using the library (built for Electron) I noticed that most of the functions (like findByIds) are not exposed. I thought there was something wrong with the build, but looking at the src/node_usb.cc file I also do not see these functions. so there is probably something very basic that I'm missing here...
Can anyone help ?

kevinmehall, 2017-02-18:
The public API is in https://github.com/tessel/node-usb/blob/master/usb.js, which calls into internal C++ functions. That public API wrapper is what you should get when you `require('usb')`, though.

daysleeper29, 2017-02-22:
Yeah, that's it, thanks. I required the .node file instead of 'usb'. Problem is, when I require 'usb' I get an 'package.json does not exists at C:\package.json' error when electron loads, besides warnings during the build process like:

WARNING in ./~/usb/usb.js
Critical dependencies:
5:37-58 the request of a dependency is an expression
 @ ./~/usb/usb.js 5:37-58

Probably an error in my webpack build script.

closed 2017-02-23 by daysleeper29