node-usb/node-usb #169: Windows 10: possible problems with combinations of nw.js, node and node-usb to obtain working noble

OwenBrotherwood, 2017-02-19:
This is a placeholder to document possible problems with combinations of nw.js node versions to obtain working noble rather than hijack other issues

At the present time, I am unable to build an nw.js application from source to obtain a working noble.

This comment gets edited as I document the issue, with possible close if I suddenly find my mistake.

https://github.com/tessel/node-usb/issues/162#issuecomment-280881267

OwenBrotherwood, 2017-02-23:
Reopened as https://github.com/tessel/node-usb/issues/162#issuecomment-280881267 is now changing to WIP with this title so as not to hijack the electron based issue.

I will be working on usb for the next couple of days and come back with a current status as time goes by.
Current status is I cannot get back to 👍 from last time I wrote with the current master.
Probably because I have not written down all the steps that I did last time or something I have overseen.

I edit this comment as time goes by to keep noise level down but indicate there is a perceived problem.

---

Next step:
- a non nw.js test setup for node and node-usb with noble.

"I'll be back"

kevinmehall, 2017-05-20:
Closing this issue for now; feel free to re-open with additional details (though I can't guarantee I can help with nw.js).

closed 2017-05-20 by kevinmehall