node-usb/node-usb #172: 'on attach' returning LIBUSB_ERROR_NOT_SUPPORTED (Win 7)

daysleeper29, 2017-02-23:
Hi, thanks for your help with my previous issue. The projects build fine now however when adding an event listener , I'm getting the 'not supported' error (I can/open close devices so the drivers are installed , although it shouldn't really matter here ). Are there any known issues on Windows 64 ?

code is pretty simple

let usb = require('usb');

usb.on('attach', function(device) { 
    console.log("Usb Device attached:\n", device);
});

Thanks

daysleeper29, 2017-02-23:
Oh, forgot I was using [Cedware's fork](https://github.com/tessel/node-usb/issues/162) (Electron issue), perhaps there's a problem there...

kevinmehall, 2017-02-23:
Yes, that fork uses upstream libusb, which does not support hotplug on Windows.

kevinmehall, 2017-02-23:
You can try using git master of this repository, which hopefully fixed the build issue that fork was intended to solve.

closed 2017-02-23 by kevinmehall