node-usb/node-usb #174: LIBUSB_ERROR_BUSY Raspberry Pi 3

victorlm125, 2017-03-07:
Firstable I had LIBUSB_ERROR_ACCESS that I fix adding permissions to usblib but now I have this error and I can't found out whats the reason because nobody uses this usb driver before my node server so I don't unsderstand.

Regards

julienvincent, 2017-03-15:
I would also like a solution to this. I am getting this on the PiB 1

julienvincent, 2017-03-15:
I did some research into this: The reason is that the pi attaches a kernel driver to connected devices. You need to check for a kernel driver and detach it before claiming the interface.

pseudo code:
```javascript
device.open()
const deviceInterface = device.interfaces[0]

let driverAttached = false
if (printerInterface.isKernelDriverActive()) {
   driverAttached = true
   deviceInterface.detachKernelDriver()
}

deviceInterface.claim()

// ... use the device

deviceInterface.release(() => {
   if (driverAttached) {
      deviceInterface.attachKernelDriver()
   }

   device.close()
})
```

kevinmehall, 2017-05-20:
:+1: We should add a "Troubleshooting FAQ" section to the README with hints like this.

kevinmehall, 2017-05-20:
Closing in favor of #182

closed 2017-05-20 by kevinmehall