node-usb/node-usb #175: Failed at the usb@1.2.0 install script..

Nic2012, 2017-03-11:
After upgarding my Raspi Zero (headless)
[2017-03-11T09_11_08_519Z-debug.txt](https://github.com/tessel/node-usb/files/835640/2017-03-11T09_11_08_519Z-debug.txt)

 and upgrading npm to the latest I cannot run npm install usb successfully:

```
root@zero0:/home/pi# npm install usb

> usb@1.2.0 install /home/pi/node_modules/usb
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download: https://github.com/tessel/node-usb/releases/download/1.2.0/usb_bindings-v1.2.0-node-v48-linux-arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for usb@1.2.0 and node@6.9.1 (node-v48 ABI) (falling back to source compile with node-gyp)
gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/6.9.1"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/node_modules/usb/.node-gyp"
make: Entering directory '/home/pi/node_modules/usb/build'
make: *** No rule to make target '../.node-gyp/6.9.1/include/node/common.gypi', needed by 'Makefile'.  Stop.
make: Leaving directory '/home/pi/node_modules/usb/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Linux 4.4.50+
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/home/pi/node_modules/usb/src/binding/usb_bindings.node" "--module_name=usb_bindings" "--module_path=/home/pi/node_modules/usb/src/binding"
gyp ERR! cwd /home/pi/node_modules/usb
gyp ERR! node -v v6.9.1
gyp ERR! node-gyp -v v3.5.0
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/pi/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/home/pi/node_modules/usb/src/binding' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/pi/node_modules/usb/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:106:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:877:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
node-pre-gyp ERR! System Linux 4.4.50+
node-pre-gyp ERR! command "/usr/local/bin/node" "/home/pi/node_modules/usb/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/pi/node_modules/usb
node-pre-gyp ERR! node -v v6.9.1
node-pre-gyp ERR! node-pre-gyp -v v0.6.30
node-pre-gyp ERR! not ok
Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/pi/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/home/pi/node_modules/usb/src/binding' (1)
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN pi No description
npm WARN pi No repository field.
npm WARN pi No README data
npm WARN pi No license field.
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! usb@1.2.0 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the usb@1.2.0 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the usb package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs usb
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls usb
npm ERR! There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2017-03-11T09_11_08_519Z-debug.log

```

kevinmehall, 2017-03-11:
Don't use NPM as root, or if you do, you apparently have to use `npm install --unsafe-perm usb`.

https://github.com/mapbox/node-pre-gyp/issues/165

Nic2012, 2017-03-13:
You are right, thank you. It works now when I install it with the pi user.
But at the beginning I still read
```
...
> usb@1.2.0 install /home/pi/node_modules/usb
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download: https://github.com/tessel/node-usb/releases/download/1.2.0/usb_bindings-v1.2.0-node-v48-linux-arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for usb@1.2.0 and node@6.10.0 (node-v48 ABI) (falling back to source compile with node-gyp)
```
But I tested the usb module, and it works fine.

closed 2017-03-13 by Nic2012