node-usb/node-usb #182: Add Troubleshooting FAQ

kevinmehall, 2017-05-20:
* LIBUSB_ERROR_BUSY
  * Linux: detach_kernel_driver (Example in #174)
* LIBUSB_ERROR_ACCESS
  * macOS: Can't open devices that have a kernel driver, no good workaround.
     * Usually comes up with HID devices -- is there a recommended libhid library?
  * Linux: permissions, udev rules
* LIBUSB_ERROR_NOT_SUPPORTED
  * Windows: Bind WinUSB driver with libwdi / Zadig
    * Mention WCID for custom devices

fatso83, 2018-03-13:
Some mention of how to enable those access rights without requiring `sudo` would be nice as well.

mrbar42, 2019-04-03:
After fighting `LIBUSB_ERROR_NOT_FOUND` on an OSX Mojave for a while i ended up using [node-hid](https://github.com/node-hid/node-hid) which worked effortlessly.
might serve as a workaround for others with HID devices.

mcuee, 2023-05-03:
> is there a recommended libhid library?

Yes, libusb project recommends HIDAPI for HID device. It supports both USB and Bluetooth HID and even other types of HID devices. There is a active node binding node-hid as well.

hidapi project:
https://github.com/libusb/hidapi

node-hid:
https://github.com/node-hid/node-hid

mcuee, 2023-05-03:
> macOS: Can't open devices that have a kernel driver, no good workaround

Please refer to this discussion, libusb-1.0.26 adds the support to detach kernel driver with correct entitlement (which is diffcult to get from Apple) but at least you can run as root.
* https://github.com/libusb/libusb/issues/1014

mcuee, 2023-05-03:
Please read libusb FAQ here.
https://github.com/libusb/libusb/wiki/FAQ

fatso83, 2023-05-04:
Better late than never 😄

PR #611 (2023-05-06): Add reference to libusb FAQ
## Changes

- Add reference to libusb FAQ, covering the majority of troubleshooting issues

## Fixes

Fixes #182 

## Checklist

Have you...

- [ ] Tested the change acts as expected
- [ ] Checked the change doesn't remove or change existing functionality
- [ ] Considered how the feature impacts the product and tested around it (not just the happy paths)
- [ ] Where possible, executed the hardware tests on multiple operating systems
closed 2023-05-06 by thegecko