node-usb/node-usb #185: USBInterfaceOpen: another process has device opened for exclusive access

SantoDE, 2017-06-29:
Hey folks,

I read trough the issues and all of that but still I'm a little confuesed :) I'm trying to connect to an RFID Reader trough CDI. The firmware of the device is flashed to CDI as well and I don't see it as a HID Device either.

I can't connect (= claim) to the "proper" interface based on the error mentioned above. My guess is, that drivers such as `com.apple.driver.usb.cdc` or `com.apple.driver.usb.cdc.acm` are just faster than I am and therefore its failing. Is there a way, to use a own configuration or so to make it possible to use the libusb driver (if that makes sense)?

Looking forward to your replies & thanks.

Regards,
SantoDE

kevinmehall, 2017-06-29:
Unfortunately on macOS there is no way to access a device with libusb / node-usb when it is claimed by a kernel driver. If your device implements CDC/ACM serial, look into the `node-serial` library to access it through the kernel's serial port driver.

SantoDE, 2017-06-30:
Yeah, that's what I'm doing right now. Thanks! :)

closed 2017-06-30 by SantoDE