node-usb/node-usb #186: How do i get the port name of a specific device?

Jeanmass, 2017-06-30:
Hey guys ! 

Thanks for this library, it's working great for me, except for one thing. I couldnt find a way to get the port name of a specific device (COMX, /dev/ttyUSB0)?  

Can it be done? Thanks !

liyatang, 2017-07-06:
try [usb-detection](https://github.com/MadLittleMods/node-usb-detection)

kevinmehall, 2017-12-01:
Correct, this library is for accessing USB devices that don't have in-kernel drivers, not usb-serial, so isn't a good fit for this use case.

closed 2017-12-01 by kevinmehall

thegecko, 2024-04-18:
I think your best bet is now https://serialport.io/

ShafSpecs, 2024-04-19:
> I think your best bet is now https://serialport.io/

That's what I was using but they don't have events so you can't create a hotplugging system. Except via smthg like intervals which is super inefficient especially when I'm not plugging/unplugging every 10 seconds. I was thinking of using this with it and cross checking the serial numbers but this doesn't seem to recognize UART devices

thegecko, 2024-04-19:
Using both is one way we do it, but it relies on all your devices having a serial number:

- node-usb hotplug event > serialnumber
- use serialnumber in serialport to find device > com port