node-usb/node-usb #189: Doesn't build on windows 10 with VS 2017

soapdog, 2017-07-10:
I can't build this on Windows 10 with Visual Studio 2017. It boils down to:

```
gyp verb cli [ 'C:\\Program Files\\nodejs\\node.exe',
gyp verb cli   'C:\\Users\\soapdog\\AppData\\Roaming\\npm\\node_modules\\t2-cli\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js',
gyp verb cli   'build',
gyp verb cli   '--fallback-to-build',
gyp verb cli   '--module=C:\\Users\\soapdog\\AppData\\Roaming\\npm\\node_modules\\t2-cli\\node_modules\\usb\\src\\binding\\usb_bindings.node',
gyp verb cli   '--module_name=usb_bindings',
gyp verb cli   '--module_path=C:\\Users\\soapdog\\AppData\\Roaming\\npm\\node_modules\\t2-cli\\node_modules\\usb\\src\\binding' ]
gyp info using node-gyp@3.6.0
gyp info using node@7.10.0 | win32 | x64
gyp verb command build []
gyp verb build type Release
gyp verb architecture x64
gyp verb node dev dir C:\Users\soapdog\.node-gyp\7.10.0
gyp verb found first Solution file build/binding.sln
gyp verb `which` succeeded for `msbuild` C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\MSBuild\15.0\bin\msbuild.EXE
gyp verb "Release" dir needed to be created? null
gyp verb copying "node.lib" for x64 C:\Users\soapdog\.node-gyp\7.10.0\Release\node.lib
gyp info spawn msbuild
gyp info spawn args [ 'build/binding.sln',
gyp info spawn args   '/clp:Verbosity=minimal',
gyp info spawn args   '/nologo',
gyp info spawn args   '/p:Configuration=Release;Platform=x64' ]
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
MSBUILD : error MSB4025: The project file could not be loaded. Root element is missing.
gyp ERR! build error
gyp ERR! stack Error: `msbuild` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Users\soapdog\AppData\Roaming\npm\node_modules\t2-cli\node_modules\npm\node_modules\node-gyp\lib\build.js:285:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:194:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Windows_NT 10.0.15063
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\soapdog\\AppData\\Roaming\\npm\\node_modules\\t2-cli\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "build" "--fallback-to-build" "--module=C:\\Users\\soapdog\\AppData\\Roaming\\npm\\node_modules\\t2-cli\\node_modules\\usb\\src\\binding\\usb_bindings.node" "--module_name=usb_bindings" "--module_path=C:\\Users\\soapdog\\AppData\\Roaming\\npm\\node_modules\\t2-cli\\node_modules\\usb\\src\\binding"gyp ERR! cwd C:\Users\soapdog\AppData\Roaming\npm\node_modules\t2-cli\node_modules\usb
gyp ERR! node -v v7.10.0
gyp ERR! node-gyp -v v3.6.0
```
Apparently the important tidbit is:

```
MSBUILD : error MSB4025: The project file could not be loaded. Root element is missing.
```

No idea how to fix but without this, **t2-cli** doesn't build on Windows 10.

drjkhoury, 2017-11-30:
I just came across this. I am sure you already solved it, but in case not, here is a thread I found today:

https://github.com/nodejs/node-gyp/issues/1144

thegecko, 2021-10-18:
Please re-test this with v1.8.0 or newer and let me know if this fixes the issue.

thegecko, 2021-11-08:
Closing due to inactivity, please reopen this issue if it persists with the latest release.

closed 2021-11-08 by thegecko