node-usb/node-usb #194: "Could not initialize libusb. Check that your system has a usb controller." on Travis CI

jviotti, 2017-08-15:
For some reason I get that error only when running inside a Travis CI job.

I tracked down the error to `libusb_init` (http://libusb.sourceforge.net/api-1.0/group__lib.html#ga9517c37281bba0b51cc62eba728be48b). Do you have any clues about why that's failing, and how we can at least workaround it?

jviotti, 2017-08-15:
We run our builds inside Docker, so maybe that's the issue?

adenvt, 2017-08-21:
Got same error on VPS too, i think  because it hasn't usb controller.

BTW, Is there any way to catch error... ?

sebakerckhof, 2018-01-08:
same issue here, it's indeed because it doesn't have a usb controller

nickmetal, 2018-01-25:
Have same problem with using Windows(10). It fired when I just running webpack devserver without using any usb devices. With OS X and Linux work well

agirorn, 2018-07-30:
Why does this module throw this error here [usb.js#L11](https://github.com/tessel/node-usb/blob/12aee6d773fa75134a6757839d4f0e254e3d18fb/usb.js#L11) this means that every code that requires this module needs to guard it is self against this (bug I think) even if it is not going to use the USB, and just wants go require it in the top of the js file and then later on do some feature detection.

Since the property `INIT_ERROR` is already on the module it should be enough to silently ignore this and allow the user of the module to decide how he want's to handle this scenario.

Would you accept a pull request that removes this check and adds the available function a simple and clean API to check if USB has been detected?
```js
usb.available = () => !usb.INIT_ERROR;
```

kevinmehall, 2018-07-30:
@agirorn If you want to make a PR removing this `throw`, I think the better API would be to check whether initialization failed in `getDeviceList` and anything else that can be called without a device, and make those functions throw an exception.

webmaster128, 2018-08-17:
I'd be happy for feedback on the way @kevinmehall described implemented in #260

PR #260 (2018-08-19): Allow loading module in environments without usb
e.g. CI system.

This throws an exception when USB is not available only when used, not when module is loaded.

Check currently happens

* in findByIds()
* in Device.open()
* in Interface constructor
* in Endpoint constructor

Please let me know if those places should be reduced or expanded. I am not an expert of USB.

Closes #194
commit 5158dc0b (2018-08-19): Allow loading module in environments without usb

e.g. CI system. Closes #194
closed 2018-08-19 by kevinmehall