node-usb/node-usb #195: issues with bash on ubuntu on windows

rachelnicole, 2017-08-17:
Tried installing t2-cli, and ran into some issues. Not exactly sure what's happening but @HipsterBrown asked me to file an issue. Here's my log:

```
rachel ✨  ➜  face ➜ [master !?]: npm install t2-cli -g
/usr/bin/t2 -> /usr/lib/node_modules/t2-cli/bin/tessel-2.js
/usr/bin/cargo-tessel -> /usr/lib/node_modules/t2-cli/bin/cargo-tessel.js

> usb@1.2.0 install /usr/lib/node_modules/t2-cli/node_modules/usb
> node-pre-gyp install --fallback-to-build

node-pre-gyp info it worked if it ends with ok
node-pre-gyp verb cli [ '/usr/bin/nodejs',
node-pre-gyp verb cli   '/usr/lib/node_modules/t2-cli/node_modules/usb/node_modules/.bin/node-pre-gyp',
node-pre-gyp verb cli   'install',
node-pre-gyp verb cli   '--fallback-to-build' ]
node-pre-gyp info using node-pre-gyp@0.6.30
node-pre-gyp info using node@8.1.3 | linux | x64
node-pre-gyp verb command install []
node-pre-gyp info check checked for "/usr/lib/node_modules/t2-cli/node_modules/usb/src/binding/usb_bindings.node" (not found)
node-pre-gyp http GET https://github.com/tessel/node-usb/releases/download/1.2.0/usb_bindings-v1.2.0-node-v57-linux-x64.tar.gz
node-pre-gyp http 404 https://github.com/tessel/node-usb/releases/download/1.2.0/usb_bindings-v1.2.0-node-v57-linux-x64.tar.gz
node-pre-gyp ERR! Tried to download: https://github.com/tessel/node-usb/releases/download/1.2.0/usb_bindings-v1.2.0-node-v57-linux-x64.t
ar.gz
node-pre-gyp ERR! Pre-built binaries not found for usb@1.2.0 and node@8.1.3 (node-v57 ABI) (falling back to source compile with node-gyp
)
node-pre-gyp http Pre-built binary not available for your system, looked for https://github.com/tessel/node-usb/releases/download/1.2.0/
usb_bindings-v1.2.0-node-v57-linux-x64.tar.gz
node-pre-gyp verb command build [ 'rebuild' ]
gyp ERR! configure error
gyp ERR! stack Error: Can't find Python executable "python", you can set the PYTHON env variable.
gyp ERR! stack     at PythonFinder.failNoPython (/usr/lib/node_modules/node-gyp/lib/configure.js:483:19)
gyp ERR! stack     at PythonFinder.<anonymous> (/usr/lib/node_modules/node-gyp/lib/configure.js:397:16)
gyp ERR! stack     at F (/usr/lib/node_modules/node-gyp/node_modules/which/which.js:68:16)
gyp ERR! stack     at E (/usr/lib/node_modules/node-gyp/node_modules/which/which.js:80:29)
gyp ERR! stack     at /usr/lib/node_modules/node-gyp/node_modules/which/which.js:89:16
gyp ERR! stack     at /usr/lib/node_modules/node-gyp/node_modules/isexe/index.js:42:5
gyp ERR! stack     at /usr/lib/node_modules/node-gyp/node_modules/isexe/mode.js:8:5
gyp ERR! stack     at FSReqWrap.oncomplete (fs.js:152:21)
gyp ERR! System Linux 4.4.0-43-Microsoft
gyp ERR! command "/usr/bin/nodejs" "/usr/lib/node_modules/node-gyp/bin/node-gyp.js" "configure" "--fallback-to-build" "--module=/usr/lib
/node_modules/t2-cli/node_modules/usb/src/binding/usb_bindings.node" "--module_name=usb_bindings" "--module_path=/usr/lib/node_modules/t
2-cli/node_modules/usb/src/binding"
gyp ERR! cwd /usr/lib/node_modules/t2-cli/node_modules/usb
gyp ERR! node -v v8.1.3
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute '/usr/bin/nodejs /usr/lib/node_modules/node-gyp/bin/node-gyp.js configure --fallback-to
-build --module=/usr/lib/node_modules/t2-cli/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/us
r/lib/node_modules/t2-cli/node_modules/usb/src/binding' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/t2-cli/node_modules/usb/node_modules/node-pre-gyp/lib/uti
l/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:125:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:213:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:897:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:208:5)
node-pre-gyp ERR! System Linux 4.4.0-43-Microsoft
node-pre-gyp ERR! command "/usr/bin/nodejs" "/usr/lib/node_modules/t2-cli/node_modules/usb/node_modules/.bin/node-pre-gyp" "install" "--
fallback-to-build"
node-pre-gyp ERR! cwd /usr/lib/node_modules/t2-cli/node_modules/usb
node-pre-gyp ERR! node -v v8.1.3
node-pre-gyp ERR! node-pre-gyp -v v0.6.30
node-pre-gyp ERR! not ok
Failed to execute '/usr/bin/nodejs /usr/lib/node_modules/node-gyp/bin/node-gyp.js configure --fallback-to-build --module=/usr/lib/node_m
odules/t2-cli/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/usr/lib/node_modules/t2-cli/node_
modules/usb/src/binding' (1)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! usb@1.2.0 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the usb@1.2.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2017-08-17T17_37_35_962Z-debug.log
```

rwaldron, 2017-08-17:
> gyp ERR! stack Error: Can't find Python executable "python", you can set the PYTHON env variable.

Can you confirm that Python is installed and that the path is set? Thanks!

kevinmehall, 2017-08-17:
As far as I know, WSL doesn't support libusb (doesn't implement usbfs), so it likely won't work even if you do get it to build.

closed 2017-09-23 by kevinmehall