node-usb/node-usb #200: Crashes when running on OSx

juanmas07, 2017-10-23:
```
  var usb = require('usb');
  var devices = usb.getDeviceList();

```
```

dyld: lazy symbol binding failed: Symbol not found: __ZN2v86Object8ForceSetENS_5LocalINS_5ValueEEES3_NS_17PropertyAttributeE
  Referenced from: /Users/juan/Documents/manu/dev/sw/recargas-desktop/node_modules/usb/src/binding/usb_bindings.node
  Expected in: flat namespace

dyld: Symbol not found: __ZN2v86Object8ForceSetENS_5LocalINS_5ValueEEES3_NS_17PropertyAttributeE
  Referenced from: /Users/juan/Documents/manu/dev/sw/recargas-desktop/node_modules/usb/src/binding/usb_bindings.node
  Expected in: flat namespace
```

```
brew config
HOMEBREW_VERSION: 1.3.6
ORIGIN: https://github.com/Homebrew/brew
HEAD: 270b752f5d9d218bfbed6fe85b6974fa653fb25f
Last commit: 5 days ago
Core tap ORIGIN: https://github.com/Homebrew/homebrew-core
Core tap HEAD: b7d1e926e9f2e3e950866b7e82ad1fbe34f1fb12
Core tap last commit: 4 hours ago
HOMEBREW_PREFIX: /usr/local
HOMEBREW_REPOSITORY: /usr/local/Homebrew
HOMEBREW_CELLAR: /usr/local/Cellar
HOMEBREW_BOTTLE_DOMAIN: https://homebrew.bintray.com
CPU: quad-core 64-bit broadwell
Homebrew Ruby: 2.3.3 => /usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.3.3/bin/ruby
Clang: 9.0 build 900
Git: 2.12.2 => /usr/local/bin/git
Perl: /usr/bin/perl
Python: /usr/local/bin/python => /usr/local/Cellar/python/2.7.12_2/Frameworks/Python.framework/Versions/2.7/bin/python2.7
Ruby: /usr/bin/ruby => /System/Library/Frameworks/Ruby.framework/Versions/2.0/usr/bin/ruby
Java: 1.8.0_91
macOS: 10.12.6-x86_64
Xcode: N/A
CLT: 9.0.1.0.1.1506734476
X11: 2.7.9 => /opt/X11
```

juanmas07, 2017-10-25:
Even with xcode installed

```
brew config
HOMEBREW_VERSION: 1.3.6
ORIGIN: https://github.com/Homebrew/brew
HEAD: 270b752f5d9d218bfbed6fe85b6974fa653fb25f
Last commit: 7 days ago
Core tap ORIGIN: https://github.com/Homebrew/homebrew-core
Core tap HEAD: b7d1e926e9f2e3e950866b7e82ad1fbe34f1fb12
Core tap last commit: 2 days ago
HOMEBREW_PREFIX: /usr/local
HOMEBREW_REPOSITORY: /usr/local/Homebrew
HOMEBREW_CELLAR: /usr/local/Cellar
HOMEBREW_BOTTLE_DOMAIN: https://homebrew.bintray.com
CPU: quad-core 64-bit broadwell
Homebrew Ruby: 2.3.3 => /usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.3.3/bin/ruby
Clang: 9.0 build 900
Git: 2.12.2 => /usr/local/bin/git
Perl: /usr/bin/perl
Python: /usr/local/bin/python => /usr/local/Cellar/python/2.7.12_2/Frameworks/Python.framework/Versions/2.7/bin/python2.7
Ruby: /usr/bin/ruby => /System/Library/Frameworks/Ruby.framework/Versions/2.0/usr/bin/ruby
Java: 1.8.0_91
macOS: 10.12.6-x86_64
Xcode: 9.0.1
CLT: 9.0.1.0.1.1506734476
X11: 2.7.9 => /opt/X11
```

kevinmehall, 2017-12-01:
What Node version are you using? This looks like the module is compiled against the wrong version of V8. Typically Node's ABI version check should prevent this from even loading.  Try rebuilding from source (`npm install usb --build-from-source`).

This library bundles a copy of libusb and isn't using the version from brew.

thegecko, 2021-11-08:
Closing due to inactivity, please reopen this issue if it persists with the latest release.

closed 2021-11-08 by thegecko