node-usb/node-usb #211: V1.4 Interface.claim() fails with LIBUSB_ERROR_NO_DEVICE using USB 1.1 device

rhuehn, 2017-12-07:
I upgraded from v1.2->v1.4 to fix build errors on node v9 and had the bad luck to find node-usb broke with my X10 CM15A device. A temporary workaround theatersoft/x10@59470a8 pins the working version before #209 PR.

Probable cause is a missed edge case with devices lacking a Binary Object Store descriptor, as my problem device is USB 1.1.

kevinmehall, 2017-12-07:
I've in fact only tested it with devices _without_ a BOS descriptor, so I don't think that's solely the problem.

What OS are you on, and can you post a libusb debug log by setting the LIBUSB_DEBUG environment variable before running the program?
Linux or OSX: `export LIBUSB_DEBUG=4`
Windows: `set LIBUSB_DEBUG=4`

rhuehn, 2017-12-07:
Ubuntu 17.10.

Good (before BOS commit) log:
```[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000002] [00007ec1] libusb: debug [libusb_init] created default context
[ 0.000018] [00007ec1] libusb: debug [libusb_init] libusb v1.0.19.10914
[ 0.000032] [00007ec1] libusb: debug [find_usbfs_path] found usbfs at /dev/bus/usb
[ 0.000038] [00007ec1] libusb: debug [op_init] bulk continuation flag supported
[ 0.000041] [00007ec1] libusb: debug [op_init] zero length packet flag supported
[ 0.000047] [00007ec1] libusb: debug [op_init] sysfs can relate devices
[ 0.000049] [00007ec1] libusb: debug [op_init] sysfs has complete descriptors
[ 0.000140] [00007ecc] libusb: debug [linux_udev_event_thread_main] udev event thread entering.
[ 0.001445] [00007ec1] libusb: debug [linux_get_device_address] getting address for device: usb3 detached: 0
[ 0.001450] [00007ec1] libusb: debug [linux_get_device_address] scan usb3
[ 0.001470] [00007ec1] libusb: debug [linux_get_device_address] bus=3 dev=1
[ 0.001473] [00007ec1] libusb: debug [linux_enumerate_device] busnum 3 devaddr 1 session_id 769
[ 0.001476] [00007ec1] libusb: debug [linux_enumerate_device] allocating new device for 3/1 (session 769)
[ 0.001546] [00007ec1] libusb: debug [linux_get_device_address] getting address for device: 3-1 detached: 0
[ 0.001550] [00007ec1] libusb: debug [linux_get_device_address] scan 3-1
[ 0.001564] [00007ec1] libusb: debug [linux_get_device_address] bus=3 dev=2
[ 0.001566] [00007ec1] libusb: debug [linux_enumerate_device] busnum 3 devaddr 2 session_id 770
[ 0.001568] [00007ec1] libusb: debug [linux_enumerate_device] allocating new device for 3/2 (session 770)
[ 0.001587] [00007ec1] libusb: debug [linux_get_parent_info] Dev 0x3123a80 (3-1) has parent 0x330c750 (usb3) port 1
[ 0.001631] [00007ec1] libusb: debug [linux_get_device_address] getting address for device: 3-2 detached: 0
[ 0.001633] [00007ec1] libusb: debug [linux_get_device_address] scan 3-2
[ 0.001646] [00007ec1] libusb: debug [linux_get_device_address] bus=3 dev=3
[ 0.001648] [00007ec1] libusb: debug [linux_enumerate_device] busnum 3 devaddr 3 session_id 771
[ 0.001650] [00007ec1] libusb: debug [linux_enumerate_device] allocating new device for 3/3 (session 771)
[ 0.001663] [00007ec1] libusb: debug [linux_get_parent_info] Dev 0x3123b80 (3-2) has parent 0x330c750 (usb3) port 2
[ 0.001715] [00007ec1] libusb: debug [linux_get_device_address] getting address for device: 3-3 detached: 0
[ 0.001719] [00007ec1] libusb: debug [linux_get_device_address] scan 3-3
[ 0.001733] [00007ec1] libusb: debug [linux_get_device_address] bus=3 dev=4
[ 0.001735] [00007ec1] libusb: debug [linux_enumerate_device] busnum 3 devaddr 4 session_id 772
[ 0.001738] [00007ec1] libusb: debug [linux_enumerate_device] allocating new device for 3/4 (session 772)
[ 0.001749] [00007ec1] libusb: debug [linux_get_parent_info] Dev 0x331ed10 (3-3) has parent 0x330c750 (usb3) port 3
[ 0.001789] [00007ec1] libusb: debug [linux_get_device_address] getting address for device: 3-4 detached: 0
[ 0.001791] [00007ec1] libusb: debug [linux_get_device_address] scan 3-4
[ 0.001804] [00007ec1] libusb: debug [linux_get_device_address] bus=3 dev=5
[ 0.001806] [00007ec1] libusb: debug [linux_enumerate_device] busnum 3 devaddr 5 session_id 773
[ 0.001809] [00007ec1] libusb: debug [linux_enumerate_device] allocating new device for 3/5 (session 773)
[ 0.001823] [00007ec1] libusb: debug [linux_get_parent_info] Dev 0x3319e20 (3-4) has parent 0x330c750 (usb3) port 4
[ 0.001864] [00007ec1] libusb: debug [linux_get_device_address] getting address for device: 3-5 detached: 0
[ 0.001867] [00007ec1] libusb: debug [linux_get_device_address] scan 3-5
[ 0.001879] [00007ec1] libusb: debug [linux_get_device_address] bus=3 dev=6
[ 0.001881] [00007ec1] libusb: debug [linux_enumerate_device] busnum 3 devaddr 6 session_id 774
[ 0.001883] [00007ec1] libusb: debug [linux_enumerate_device] allocating new device for 3/6 (session 774)
[ 0.001896] [00007ec1] libusb: debug [linux_get_parent_info] Dev 0x331a770 (3-5) has parent 0x330c750 (usb3) port 5
[ 0.001921] [00007ec1] libusb: debug [linux_get_device_address] getting address for device: 3-5.2 detached: 0
[ 0.001923] [00007ec1] libusb: debug [linux_get_device_address] scan 3-5.2
[ 0.001937] [00007ec1] libusb: debug [linux_get_device_address] bus=3 dev=8
[ 0.001939] [00007ec1] libusb: debug [linux_enumerate_device] busnum 3 devaddr 8 session_id 776
[ 0.001941] [00007ec1] libusb: debug [linux_enumerate_device] allocating new device for 3/8 (session 776)
[ 0.001951] [00007ec1] libusb: debug [linux_get_parent_info] Dev 0x3224a10 (3-5.2) has parent 0x331a770 (3-5) port 2
[ 0.002004] [00007ec1] libusb: debug [linux_get_device_address] getting address for device: 3-5.3 detached: 0
[ 0.002006] [00007ec1] libusb: debug [linux_get_device_address] scan 3-5.3
[ 0.002018] [00007ec1] libusb: debug [linux_get_device_address] bus=3 dev=10
[ 0.002020] [00007ec1] libusb: debug [linux_enumerate_device] busnum 3 devaddr 10 session_id 778
[ 0.002022] [00007ec1] libusb: debug [linux_enumerate_device] allocating new device for 3/10 (session 778)
[ 0.002033] [00007ec1] libusb: debug [linux_get_parent_info] Dev 0x31238d0 (3-5.3) has parent 0x331a770 (3-5) port 3
[ 0.002089] [00007ec1] libusb: debug [linux_get_device_address] getting address for device: 3-7 detached: 0
[... cut ...]
[ 0.509521] [00000278] libusb: debug [handle_events] poll() returned 1
[ 0.509524] [00000278] libusb: debug [handle_events] caught a fish on the hotplug pipe
[ 0.509528] [00000278] libusb: debug [handle_events] poll() 3 fds with timeout in 0ms
[ 0.509531] [00000278] libusb: debug [handle_events] poll() returned 0
[ 0.509534] [00000278] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.509536] [00000278] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
```

kevinmehall, 2017-12-08:
So the error is indeed "correct" in that the device is disconnecting from the bus. It's possible that the request for the BOS descriptor is crashing the device firmware and causing it to reset. 

I didn't review that BOS change carefully enough and assumed that `libusb_get_bos_descriptor` was returning cached data from the kernel like `libusb_get_config_descriptor`, but this shows it making requests to the device. Firmware bugs aside, this library should not be making (blocking!) bus traffic that you didn't ask for. So I think I'll revert that change.

closed 2023-02-26 by thegecko