node-usb/node-usb #212: How to get a device description (model, manufacturer) out of the data from node-usb?

mpathy, 2017-12-08:
First of all, this plugin is PERFECT for our needs, as we create a monitoring agent now. Especially with the events provided.

My question would be, if there is a possibility to get the device description somehow?

The different platforms all have to possibilty to show the correct names of the model and manufacturer, can I access them somehow.

Or, otherwise, is there some API out in the internet to request information based on the ids?

mpathy, 2017-12-08:
I found this file: http://www.linux-usb.org/usb.ids - which is my best guess. Maybe you can include it into the module, but to ask the system directly will get better information for sure?

kevinmehall, 2017-12-08:
See https://github.com/tcr/node-lsusb/blob/master/index.js for an example of asking the device for its manufacturer and product string descriptors.

Do note that the normal Linux `lsusb` command looks up the VID and PID in the usb.ids list rather than reading the string descriptors. These approaches may return different results, but neither is always better than the other.

closed 2017-12-08 by kevinmehall